dvmarinoff/Auuki #1: Speed in ERG mode

dvmarinoff, 2020-10-25:
The Fitness Machine Equipment Service (bluetooth spec) calculates the speed of the flywheel and sends this as a speed value.
In SIM mode this is fine, but in ERG mode this makes the speed values meaningless. In ERG mode you are setting a power target, which the trainer needs to meet regardless of gear combination or cadence, which results in power output having no effect on speed.
You can set the bike in 34/17 gear and doing 300 watts or 100 watts have no effect on speed, on the other hand changing cadence from 70 to 90 will give you about 6km/h more. This results in wrong distance calculation.

The problem:
1. If avg speed is not that important, distance is. It is a major metric for comparing time blocks of training.
Many people wouldn't like seeing their monthly distance graph sink in (especially on strava).
2. For most trainers using the small ring results in much less noise. The mid tier trainers are basically silent at low flywheel speeds.
3. The  

The solution:
Assuming flat terrain, the same way default SIM mode does, ERG speed can be calculated from power output.

The challenges:
This will require some assumptions to be made about:
- Aero drag from rider position
- Aero drag from equipment
- Air pressure
- Altitude
- Temperature
- Dew point
- Wind Speed and direction
- Surface rolling resistance
- Tires rolling resistance
- Bike weight
- ...
- (curvature of expanding spacetime =) )

The model could be simplified. At the end the resulting distance needs to match expectations (zwift, tr, ...) more than physics.

mjunkmyjunk, 2021-07-17:
did you take a look a this mathematical model from this site? You use Power to convert it down to Speed utilising the variables of slope / Air resistance and gravity.
https://www.gribble.org/cycling/power_v_speed.html

You can also see the code from 
https://github.com/WouterJD/FortiusANT/blob/ec1988dde777a7c484b8d3580748af6c333eb02b/pythoncode/usbTrainer.py#L811

or this https://physics.stackexchange.com/q/226854/287855

dvmarinoff, 2021-07-18:
Thanks, I new the gribble calc, but not the other 2.

This thread on [zwift forums](https://forums.zwift.com/t/how-should-zwift-calculate-rider-speed/547828/2)  has some useful links and this article on [slowtwitch](https://www.slowtwitch.com/Tech/The_Physics_of_Moving_a_Bike_163.html) is a nice explainer.

Overall I have decided to go in the direction of gamifying workouts. I have done some experimentation with basic 3d graphics in the browser and I think it just might work. After I finish with the basics like bluetooth, ANT+, .FIT etc and bring device support to a good level I will focus on this.

mjunkmyjunk, 2021-07-18:
I briefly scanned thru the 2 links you provided, haven't read it in-depth. There's also discussion about this by some of the Golden Cheetah developers (https://github.com/WouterJD/FortiusANT/issues/175)

In terms of device support, Would you know how to support the cycleops Hammer H1? (This is the 1st gen which uses the wahoo extension to cycling power utilising this UUID A026E005-0A7D-4AB3-97FA-F1500F9FEB8B. 

I tried using some of the logic/code from here (https://github.com/codeinversion/sensors-swift-trainers/blob/master/Sources/SwiftySensorsTrainers/WahooTrainerCharacteristic.swift) to connect and control the resistance/erg but it's not sticking. I suspect that it needs to "unlock" or "request control" but the codes are not correct or something else is happening (and I don't know what)

closed 2025-01-07 by dvmarinoff