dvmarinoff/Auuki #8: persistence for user added workouts

dvmarinoff, 2020-11-24:
The app will work with .zwo workout format only.
- [ ] research IndexDB,
- [ ] research Web Workers
- [ ] figure out how to store and work with user added collections of workouts
- [ ] research options for config files, user friendly way to create, edit and import them
- [ ] implement the solution

steffidortmund, 2021-11-23:
Hi, I really like your app!
I have just tried to import a .zwo file and every time the cool down is the other way around... is there something wrong?
[image]

dvmarinoff, 2021-11-23:
Hi, can you post here the xml content of the .zwo file?

I created a test workout similar to yours with [zwofactory](https://zwofactory.com/) and it seem alright:

[image]

Here is the content of mine:
```
<workout_file>
    <author></author>
    <name>test interval order</name>
    <description>test interval order</description>
    <sportType>bike</sportType>
    <tags>
    </tags>
    <workout>
        <Warmup Duration="600" PowerLow="0.25" PowerHigh="0.75"/>
        <IntervalsT Repeat="3" OnDuration="60" OffDuration="300" OnPower="1.21" OffPower="0.5"/>
    </workout>
</workout_file>
```

closed 2025-01-07 by dvmarinoff