dvmarinoff/Auuki #31: add workout generator

dvmarinoff, 2021-02-26:
It has to make a suggestion for a workout based on user defined constraints like time, intensity, energy system. A current physical condition and recent workout history can be used to narrow down and simplify the process for the user.
Heart rate data (like HRV) can be used to dial in intensity and energy system suggestions.
L-system or GA can be used for the generation.

fellrnr, 2022-01-31:
[image]
I've started to create a workout generator if anyone is interested. It's a simple C# desktop app.

dvmarinoff, 2022-01-31:
Curious to see this one, I’d really wish it was otherwise but I doubt that I’ll have time to work on this feature.  
Does it work on MacOS or it’s strictly Windows?

fellrnr, 2022-02-01:
Right now it's a simple windows C# app, so no Mac support. I developed it for personal use to create Golden Cheetah QwkCode statements, then did a mod to generate ZWO format with your slope extension. I'm happy to send you the code, or work on it independently and publish via Github, though I'm not going to have time to do much in the next 6 weeks.

dvmarinoff, 2022-09-22:
@lawrencemaroun
Opened a new branch for this feature: https://github.com/dvmarinoff/Flux/tree/workout-generator
and new deployment: https://flux-generator.vercel.app/