dvmarinoff/Auuki #34: Integrate 4iiii fliiiight trainer with missing 0x180A Device Information

muCruiser, 2021-04-02:
Thanks for the great app for using smart trainers. Following topic with my trainer [4iiii fliiiight](https://www.dcrainmaker.com/2019/12/4iiii-fliiiight-smart-trainer-in-depth-review.html)

The trainer is not reporting the uuid:

```
    deviceInformation: {
        uuid: '0000180a-0000-1000-8000-00805f9b34fb'

```

but instead reporting generic access:

[image]

unfortunately due to this the actual app won't work because it misses device info.

I solved this by exchanging the uuids in services.js

```
    deviceInformation: {
//new        
		uuid: '00001800-0000-1000-8000-00805f9b34fb',
        manufacturerNameString: {uuid: '00002a00-0000-1000-8000-00805f9b34fb'},
        modelNumberString:      {uuid: '00002a00-0000-1000-8000-00805f9b34fb'},
        firmwareRevisionString: {uuid: '00002a00-0000-1000-8000-00805f9b34fb'}
	}
	
// old
//		uuid: '0000180a-0000-1000-8000-00805f9b34fb',	
//        manufacturerNameString: {uuid: '00002a29-0000-1000-8000-00805f9b34fb'},
//        modelNumberString:      {uuid: '00002a24-0000-1000-8000-00805f9b34fb'},
//        firmwareRevisionString: {uuid: '00002a26-0000-1000-8000-00805f9b34fb'}
//    }
```
i did not find any manufacturerNameString, modelNumberString, firmwareRevisionString inside my ble reportings yet, so i just linked them to the device name as well

i also changed sensor location:

```
		sensorLocation:           {uuid: '00002a5D-0000-1000-8000-00805f9b34fb'},
//        sensorLocation:           {uuid: '00002a5A-0000-1000-8000-00805f9b34fb'},
```

due to ble reportings of my device and [uuid pdf](https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf)

with these changes your app was working without problems for my trainer as well.

So could you please add a routine which checks for 0x180A Device Information and if it does not find it, then checks for 0x1800 Generic Access to receive the name. Or works as well if there is no name received in device info and reports it just as "unknown"

Thanks

dvmarinoff, 2021-04-02:
Hey, thanks for the help, I'll investigate this issue and then I'll apply a solution.

muCruiser, 2021-04-04:
Thanks for the quick fix. It works and Flux can control my trainer now (with name unknown). You can add 4iiii fliiiight to your tested Trainers.

closed 2021-05-12 by dvmarinoff