dvmarinoff/Auuki #35: Add workout video view

muCruiser, 2021-04-02:
On youtube there are workout videos for training, so we could add an option to play a youtube video inside the app.
For example exchange the workout graph with something like:

```
<iframe id="ytplayer" type="text/html" width="720" height="405"
src="https://www.youtube.com/embed/ZiGE3-L4vyg"
frameborder="0" allowfullscreen>
```

Perhaps adding big buttons to change watt/resistance/slope

dvmarinoff, 2021-04-03:
I've been thinking about this use case, but I am kinda unwilling to put youtube or anything in particular inside the app itself,
mostly because different people would like to use different services like Youtube, Netflix, GCN+, or maybe run Zwift.

I see two use cases differentiated by device type:

**1. Laptops, Desktops and Tablets**

In those systems just using windows of different sizes is enough.
I am thinking of adding something like a docking mode for Flux that optimizes for this case.
So you can set the app as a short, but wide window at the top or at the bottom of the main screen and
then run Netflix, Youtube, etc above or below.

**2. Phones**

In portrait mode the screen is just too small for a whole video, and Landscape is unusable.
I can add a button that adds a video embed and then rearranges the UI elements in an optimal way.
But space is limited and it's going to be hard to make this working without removing functionality.
The only proper solution I see is extending the screen with a monitor by Screen Mirroring or Desktop mode.

**2.1 Screen mirroring**

connecting the phone to a monitor or TV with a usb-c cable, hdmi, or in worst case casting the screen

**2.2 Desktop mode**

It's the best solution, but you need a usb-c monitor (or TV).

It's available since Android 10 and many OEMs have some kind of custom solution that builds on top of it.
Like Samsung DeX, Huawei desktop mode, Motorola's Ready For, etc

- it can run many apps at the same time
I am using Samsung DeX to run Flux, GCN+, Zwift and Zwift Companion App, simultaneously all on a Samsung S9 phone.

- it allows to take advantage of the phone screen to run yet another app
In fact you can run Zwift or Youtube fullscreen on the Monitor and have Flux on the Phone Screen.

- it can do all of this and still charge the phone

This is example of desktop mode on Samsung with DeX:

[image]

muCruiser, 2021-04-03:
I agree to your options and while reading i realized one more for the phone approach:

**2.3 Split screen mode**
Open video in dedicated app and use split screen mode to open flux. Both share the screen. Video Player controls are native out of the video app.

closed 2021-04-05 by dvmarinoff