dvmarinoff/Auuki #41: Support for maps

splanes, 2021-04-15:
Hi, First of all, this is **absolutely jaw dropping**... 👏 👏 👏 👏 

Have you thought of adding maps support, i.e. these use cases?
- Draw a route in a map
- Choose start / destination points
- Mark slopes in the route using colors green to red (or violet)
- Mark an estimated point were the user is pedaling at.
- Calculate an ETA

Some of these features are currently supported in the official tacx training app. In order to be able to choose start / destination points, you have to export google maps directions to gpx, then reconvert the gpx file and upload it to the training app, only as a premium subscriber which is pretty annoying.

It's been a while since I last coded in nodejs, but if you need to I'm offering help to improve your tool.

Thanks in advance!

dvmarinoff, 2021-04-16:
Hi, thanks!

I have thought about something similar. My idea was to leverage Google Street View for pre-riding a race course (Race Recon mode). Getting familiar with climbs, marking where attacks could happen, looking at the finish line, etc. Having this mapping between visual memory and critical points like where the gradient is the steepest on a given climb, or where is the midpoint of this climb, might work much better than trying to recover a bunch of numbers from you memory while doing a threshold effort.

The User could load a gps and elevation track and use the power output from a trainer or power meter to move along the track with Street View updating accordingly. Similar to Tacx Training, but with the ability to jump back and fourth between a predefined set of segments of interest.

There are couple of problems though.

The biggest one being the quality of elevation and gps data.
The source of that data can be:

1. Cycling head unit (or other GPS nav like sports watch)
2. Export from a service (Strava, Komoot, Garmin Connect).

They all suffer from the same problem, because they have to work with insufficient and inaccurate information, thus having to make a lot of guess work. Ultimately they face the same trade off. You either try to 'smooth' the data at a cost of important features or keep those features at the cost of noise.

In regard to elevation data this manifests as loosing a lot of gradient information, like max gradient no longer being 12% but 9%, small hills and false flats disappearing all together, or on the other path gradient changes become unnatural and harsh, sections that will feel flat or at least steady rising in real world will now have sharp shifts like 3%, 9%, 3%, 9%, 1%.
And getting both negatives is also an option (Stagflation style).

GPLama has a couple of videos on the subject:
-  [Re-Riding a Route Indoors on a Smart Trainer - Wahoo ELEMNT Bolt](https://www.youtube.com/watch?v=qxPuFIotzKE)
-  [Re-Riding a Route Indoors on a Smart Trainer - Garmin Edge 520](https://www.youtube.com/watch?v=m_X82njeO8I)

I experimented with this for a bit and decided that at that point it's going to be really hard to make a decent feature with actual good experience. Not at least with algorithmic solution. One option could be to provide a set of hand made courses.

I have a race by the end of May (and because of constantly changing restrictions and bad weather), I may need to revive the idea and give it another try.

As for contribution, I am currently refactoring the whole thing and added a test suite. After I am done with that contribution will be actually practical. My most urgent issue would be to improve support for devices.

mjunkmyjunk, 2021-07-17:
I too struggle with importing GPX files of real outdoor rides into gpsbabel and then trying to get a Good Slope gradient but they all end up as "bad". and when I smooth them out using Kalman Filter, it becomes better but like you said, max gradient becomes lower and false flats become real flats etc.

closed 2022-09-22 by dvmarinoff

TheRealMoeder, 2023-01-12:
First of all congratulations to a very elegant and functional web application. I noticed you can load a track from a fit file and reride it.

1) Did you ever look into displaying the current position on a map?
2) would it be possible to write the reridden GPS position into the downloadable fit file after finishing the workout?