dvmarinoff/Auuki #43: Pick a better name

dvmarinoff, 2021-05-03:
Flux is the name of my trainer and was just a temp choice so I can start coding the app. But it is not a good name because it conflicts with way too many other projects that have the same name including popular frameworks for React and Julia.
The new name needs to be unique and googleable. Zwift is example for good name, Trainer Road not so much (hard to pronounce or make you remember it). I've looked at cycling words in French and English, historical names in my native language and others.
Tried a couple of online name generators, but couldn't find something I like.

I need to settle on a name so I can buy a domain and put the project on some real hosting.

mjunkmyjunk, 2021-07-17:
How about webTrainer?

dvmarinoff, 2021-07-18:
Nice, keep them coming!
But I do prefer going with a made up word like Zwift or something non english, so it naturally has it's own identity.

mjunkmyjunk, 2021-07-18:
poursuivant

per the TdF of Tete De La Course and poursuivant

mjunkmyjunk, 2021-07-21:
I found a possible solution for you.

[https://static.boredpanda.com/blog/wp-content/uploads/2020/10/technology-comics-system32-1-5f7c08e8c6778__880.jpg
](url)

flame-n, 2022-01-06:
- Vitelor (from vite, quick)
- Veloverse?

Awesome work on the app so far! Just tested the demo and I love it! <3

dvmarinoff, 2022-01-06:
@flame-n , Thanks, glad you liked the App!

TClin76, 2022-01-11:
Daken

"D"imitar
M"a"rinov 
Bi“k”e 
Pow"e"r 
Trai"n"

PhatWheZ, 2022-02-15:
OpenSpin

@dvmarinoff, im super happy i found this project. Didnt want to fork out to Zwift just yet, and this does exactly what i need from an indoor trainer aspect.

Got a race coming up that im keen to prepare for, need to setup a workout based on their datapoints
[image]

GCuser99, 2022-02-19:
I really like "OpenSpin" as suggested above. Another similar - VirtualSpin...

Zhwifty, 2022-06-05:
Veloverse is a good name,  suggests both a universe and also a versus, as in against rival apps.

Great app so far. 

Would love to see some customisation of the interface, such as ability to remove fields,I crease/decrease field sizes. Like you can do with a bike computer.

afrandsen, 2022-08-12:
Just found your repo, very nice indeed. Thanks for the great work.

A danish suggestion from me: 'Træd' or 'Traed'. Translates to 'step' or 'push'.
Fx. 'Træd i pedalerne!' means 'Step on the pedals!'.

dvmarinoff, 2022-08-15:
@afrandsen Thanks!

tread.app or something, sounds good.

closed 2024-10-28 by dvmarinoff