dvmarinoff/Auuki #50: add Support for Wahoo trainer control

dvmarinoff, 2021-07-19:
wahoo uses extended CPS for trainer control

list of known devices:
CycleOps Hammer H1

TClin76, 2021-12-23:
Is there anyone successfully pairing Wahoo Kickr Bike as controllable device?
It seems that it could not be found during device pairing in controllable, only be found in power meter.

dvmarinoff, 2021-12-23:
Wahoo has added support for bluetooth FTMS to the Kickr bike since June this year. Which makes their custom protocol obsolete. It should work with FTMS, but there is a chance something is going wrong during the connection attempt, or the connection is being taken by some other device.

```
KICKR BIKE Firmware v1.22.0 - 22 June, 2021
Added: Support for Bluetooth FTMS
```

- Does the Flux controllable indicator blink to yellow and then go to red,
- or there is just nothing in the list of discoverable devices?

I am almost done with the cadence feature, just struggling to find and configure a good enough simulator. But along with it there comes a big cleanup and optimisation of the the whole bluetooth code. I'll be pushing all of that to the development test version in a couple of hours. If you have access to the Kickr bike we can try to debug this.

TClin76, 2021-12-23:
KICKR BIKE: just nothing in the list of discoverable devices

Cadence feature: I found this feature in development version this morning. I will take a try when I get back to home.
Before the real test, I just think about what would be happened if both CONTROLLABLE and SPEED AND CADENCE are connected. 
APP will receive two cadence data, how does the APP know which one it should use?

dvmarinoff, 2021-12-23:
I've added just the UI for speed and cadence, the bluetooth code is in testing, maybe another hour or so and I'll push it.

Else in the connection table you will notice that speed and cadence can be come from bluetooth controllable, power meter or speed and cadence sensor, some of the values are in white colour while others are in grey. If you have say controllable and cadence sensor connected, you can tap on a source to switch it on and off, you are still connected to the device, but the app will ignore the incoming data if the source is in grey. This way you can mix and match sources of data between the connected devices.

[image]

Note: I've made the code to auto switch to reading power from a power meter when one is connected, but the others require a manual choice.

TClin76, 2021-12-23:
Noted!! It is a perfect way for manually choosing source of data.

TClin76, 2021-12-23:
Cadence is connected sucessfully (Bluetooth indicator is in green color), however the value stay in "0" while pedaling.
Another issue is that my smart trainer is now unable to be connected in Controllable for this development version 
(Bluetooth indicator for controllable blink to yellow and then go to red).
For your reference.

dvmarinoff, 2021-12-23:
Thanks for the feedback, I am still working on it.

- Cadence is now tested against some data derived from the spec description, should work now, but still it is untested with real device.  It should read sensible data, but for some edge cases it might show strange numbers.
- currently working on some improvements for FE-C over BLE, so if you have a trainer using that protocol (maybe a Tacx trainer), it could be broken on development until I finish with that.
- also earlier today the connectivity UI was broken on development for all but speed and cadence, I was in the process of redesigning it, but now it works.
- I am still investigating the case for the Kickr Bike, no idea why it isn't showing up, but I collected some data, need time to look into it.

A couple of hard page refreshes should bring the latest version of development.

As of now I expect FE-C over BLE to be only partially working as well as cadence.

TClin76, 2021-12-23:
Trainer is now connected as Controllable, but no value.
Cadence is now showing some value, and then soon jump to 0. Keep repeating like this.

BTW, my trainer is Magene Gravat.
My friend owns a Kickr bike so he can help to test it.

dvmarinoff, 2021-12-23:
finished with FE-C over BLE, confirmed to fully work, latest development version is 0.1.1.

TClin76, 2021-12-23:
Controllable works fine in version 0.1.1 now. Thanks for the prompt fix.
Cadence still has the same problem ( rpm -> 0 -> rpm -> 0 .......... )

UnJourAvec, 2021-12-28:
I have been following the project with a lot of interest. Myself I own a Kickr snap, it contains the wahoo characteristic a026e005-0a7d-4ab3-97fa-f1500f9feb8b in the 00001818-0000-1000-8000-00805f9b34fb service (cycling power). However, trainer is not recognized in the flux app. I have been playing around a bit for myself and unlocking the device by writing 0x00 into wahoo characteristic does not work to control the trainer. Any ideas here?

dvmarinoff, 2021-12-29:
Hi @UnJourAvec,
I started working on this. I did some changes to the BLE scan filters and the latest development version
0.1.2 at [flux-devel.vercel.app](https://flux-devel.vercel.app/) should be able to at least recognise the Wahoo KICKR Snap and connect to it. I also started a very basic implementation should read data and set power target, but I didn't have time to test it against a simulator today so it is very unlikely to work.

Since there isn’t a public spec for Wahoo trainer control I am using this project [sensors-swift-trainers](https://github.com/codeinversion/sensors-swift-trainers/blob/master/Sources/SwiftySensorsTrainers/WahooTrainerCharacteristic.swift) as a guideline. I don’t know how up to date it is, but it’s the only source I have.

By looking at the code to unlock the trainer I reckon you would need to subscribe for indications from the Wahoo Trainer Characteristic, and then send unlock command, which should be: `[0x20, 0xEE, 0xFC]`, and an ERG mode command that sets the power target to 200 Watts would be `[0x42, 0xC8, 0x00]`.

Can you try these operations with nRF Connect for Android or something similar and post here some logs, it would be great help. Later I'll try to write down a manual on how to work with nRF Connect, let me know if you need help.

UnJourAvec, 2021-12-29:
Hi @dvmarinoff ,

Thanks for your response. First, I checked with the 0.1.2 version. The controller is not recognized by the web application. Myself I scan for 'cycling_power', (service uuid 00001818-0000-1000-8000-00805f9b34fb) to find the snap, this does work.

I tried to unlock with  0x20 and it worked and than write with 0x42. Where did you find this information? I have been looking all over the internet to find the unlocking code but did not find a working one. 

I tested it in google chrome web app. Here is the log file. First I connect than start writing power target, first 50 than 60. I realise the log file is no easy to read.
[Wahoo Kickr Snap test.log](https://github.com/dvmarinoff/Flux/files/7790407/Wahoo.Kickr.Snap.test.log)

So let me attach some snippets:
[image]
[image]

dvmarinoff, 2021-12-29:
`Where did you find this information?`

The op codes are in [here](https://github.com/codeinversion/sensors-swift-trainers/blob/93db8dd000e47233435a41ba061c23541ee18bb4/Sources/SwiftySensorsTrainers/WahooTrainerSerializer.swift#L25). Someone linked that project in another issue few months ago.

`Myself I scan for 'cycling_power'`

I was afraid that might be the case. It kinda breaks consistency with the scan filters. I added cycling_power as an optional service, but not a major filter one, I was hoping that the Kickr Snap would also have the Wahoo advanced FTMS, and would show up in the scan.

Now if I add cycling_power as a major filter the scan will show pure power meters along side controllable devices.
Would have to figure another way to filter for the Kickr Snap.

UnJourAvec, 2021-12-29:
> It kinda breaks consistency with the scan filters. 
I know, maybe this helps, but I believe cycling_power is the only option. 
[image]

BTW, I noticed getting the ERG control was not due to using 0x20 instead of 0x00 for getting control. It was due to using 0x42 instead of 0x05 to set target power. I am a bit confused though. In the sensors-swift-trainers repository you linked, the OpCode for unlocking is 32 and for setting ergmode is 66. 

How did you get 0x20 and 0x42? Also I have do not understand the trainers response. Succes response code should be 0x00, the response is different but still it works..

dvmarinoff, 2021-12-29:
I do not know Swift but I guessed that the op codes are in base 10, because in other places in the code the hex values are used explicitly (so 66 becomes 0x42).

Thanks for the log! The odd thing is that the Snap has that Wahoo advanced FTMS its the next to last service in the bottom with uuid: `a026ee0b-0a7d-4ab3-97fa-f1500f9feb8b`. I am confused why it doesn't show up. Anyway temporary changed the filters on the development version until I figure out a better solution.

I haven't got to implementing the response, maybe tomorrow.

TClin76, 2021-12-30:
Hi,
I am not familiar those FTMS things, but I look into the code in GoldenCheetah, there is another UUID seems to be the control service.

// Wahoo specific extension to the Cycling Power service
#define WAHOO_BRAKE_CONTROL_UUID "{A026E005-0A7D-4AB3-97FA-F1500F9FEB8B}"

https://github.com/GoldenCheetah/GoldenCheetah/blob/master/src/Train/BT40Device.cpp

TClin76, 2022-01-02:
Hi @dvmarinoff 
Good news. My friend tested the latest developing version with his Kickr bike, it works now!

dvmarinoff, 2022-01-02:
That's great!
Right now I am half way through with the implementation of the Wahoo Extended Cycling power characteristic A026E005-0A7D-4AB3-97FA-F1500F9FEB8B (that's the one you've also linked in the previous comment). I have only setting of power targets done, some of the parameters for SIM/Slope mode, and reading of power, speed, cadence.

So I am curious to see which control service is the Wahoo Kickr Bike using. Now there are 3 of them: FTMS, FE-C over BLE and Wahoo Extended CPS and when device connects the app will try to search for one of them starting from FTMS and if not found will try with the next one.

Is it possible for you to produce a screenshot log like the following one, I just need the part inside the orange rectangle, with the object extended if possible:
[image]

Thank you!

UnJourAvec, 2022-01-02:
> I do not know Swift but I guessed that the op codes are in base 10, because in other places in the code the hex values are used explicitly (so 66 becomes 0x42).

That explains! Curious for the kickr bike implementation. :)

TClin76, 2022-01-03:
Hi @dvmarinoff
[image]

GCuser99, 2022-02-02:
Hi, thanks for sharing your project on GitHub!

I gave it a go today and succeeded in pairing my 2017 Wahoo Kickr and Wahoo cadence sensor but Flux did not seem to "sense" the cadence during workout attempts (I tried all 3 modes)... If there is something on my end that I can do to help debug, then let me know (I'm not a software engineer so you will have to be specific!)...

[image]

dvmarinoff, 2022-02-02:
@GCuser99 Thanks!

On your particular issue I think it’s one of those 2 options:

Case 1: Cadence from Cadence sensor needs to be activated
You will need to click on the grey cadence label in the Speed and Cadence row at connection settings to activate reading cadence from the sensor and not the Controllable. More on that [here](https://github.com/dvmarinoff/Flux/discussions/91).

Case 2: Connection is failing
You will need to connect the sensor, activate it as data source for cadence, open developer console by pressing F12, choose the console tab, and send me a copy of the contents, drag-select the text and paste here or make a screenshot. One example of that [here](https://github.com/dvmarinoff/Flux/discussions/82#discussioncomment-1884220).

From the screenshot I see the connection to the cadence sensor is being established, and cadence source is grey.
So more likely it’s Case 1.

GCuser99, 2022-02-02:
@dvmarinoff your suspicion was correct - was issue #1 and recommended corrective action did the trick! I will give this a real try tomorrow... thanks!

dvmarinoff, 2022-02-08:
Most functions are working now. The biggest mystery that remains is the unlock command `32` and its' response: ` :rx :wcps :response 1,32,2`. What does `2` mean?

closed 2022-03-10 by dvmarinoff