dvmarinoff/Auuki #63: Add info on Firefox support

richlv, 2021-12-06:
Opened the online version with Firefox, but could not figure out how to get started.
As Firefox support is not mentioned in the documentation, couldn't figure out whether it's a support issue, or PEBKAC.
Could Firefox support state be documented?

dvmarinoff, 2021-12-06:
Sorry, about that, have missed to add Firefox in the support table, I am adding it now. In general I'll also add to the web version a popup of some sort that will notify new users if the browser is supported, but that's gonna happen when the project gets closer to first stable release. Thanks for catching that!

dvmarinoff, 2021-12-06:
Firefox would probably never have support, they've stated that they are against the web bluetooth and web serial API, and would not consider it for implementation.

richlv, 2021-12-06:
Thank you so much for the quick help, will try in Chrome then :)

dvmarinoff, 2021-12-06:
Chrome on Android, Mac, Windows 10, Ubuntu and probably other linux distros would be fine. Also most Blink derived browser have support: Edge, Opera, Samsung Internet, but not Brave.

The project has some quirks, but I am working on fixing those, and it's kinda dynamic right now. I'll try to right them down in a single document, that may be the start of a user guide. 

In addition to that there are some bugs and omissions that I am not aware of, so pointing those out is highly appreciated.

closed 2021-12-07 by dvmarinoff