dvmarinoff/Auuki #64: Schwinn IC8: pairing/operation

richlv, 2021-12-06:
In https://github.com/dvmarinoff/Flux, manufacturer/brand Schwinn is not mentioned.

Attempting to use IC8 (rebranded to 800IC later) with Chrome on Mac. Clicking on "CONTROLLABLE" scans and finds an entry "IC Bike". Selecting it and clicking "Pair" gives a brief pause, and the "CONTROLLABLE" light then switches from yellow to red.

Retrying this operation shows "IC Bike - Paired" :
[image]
Still, the indicator never changes to green, as seen in screenshots.

This equipment seems to support FTMS, as mentioned in https://github.com/ptx2/gymnasticon/issues/34 .

Am I doing something wrong here?

dvmarinoff, 2021-12-06:
Something is going wrong on connection, probably some service is missing or misbehaving in an expected way.
Can you open up the developer console with F12, attempt to connect and make a screenshot like this:

[image]

This may help me figure out what is going on.

dvmarinoff, 2021-12-06:
Investigating ... there is a lot of info in that issue you linked. Might catch the bug, but will need your help to verify if it works on the particular hardware, I just don't have access to this Spin bike.

richlv, 2021-12-06:
Searchable message and screenshot:

":ble 'Could not request ble:controllable' DOMException: No Characteristics matching UUID 00002ad8-0000-1000-8000-00805f9b34fb found in Service with UUID 00001826-0000-1000-8000-00805f9b34fb."

[image]

dvmarinoff, 2021-12-07:
Status Update:

I created a nRF Connect simulator with the same services and characteristics as the Schwinn according to the data in that issue. And I realised that the Schwinn is missing the regular control characteristics like the one showing in the bug log: 0x2ad8 or "Supported Power Range". My code just assumed that an FTMS device is also a control device, but that is not necessary, actually many device would use the compact format of the indoor bike data characteristic just to transmit all of their data on a single BLE connection: Power, Cadence, Speed and most notably Heart Rate. Even some trainers now allow to proxy an HRM strap by connecting it to the trainer. 

So I reworked the FTMS service to work with optional characteristics and also to find the index of and read heart rate data if available.

Also added in the UI an option to select the Controllable as a Heart Rate data source, you will see this in the Settings tab, where the Heart Rate label is in grey, tap on it to switch it on, it will become white and display Heart Rate if available from the Controllable.

Because this is a relatively big change to critical code it is on a separate branch that will be hosted here : [flux-schwinn.vercel.app](https://flux-schwinn.vercel.app/),  for a couple of days until I can verify that there are no regressions, then I'll merge it into master and use the regular domain.

You can use this one until then, and if you can I'd be happy to get some feedback if it works with the actual Spin Bike.

richlv, 2021-12-07:
Wonderful, thank you - it now connects successfully, and gets some values (power/cadence/speed).
Will try it out some more later.

dvmarinoff, 2021-12-07:
In the pictures it looks like the spin bike has a manual resistance knob.

richlv, 2021-12-07:
It does, but it's magnetic resistance - wouldn't that be controllable?

dvmarinoff, 2021-12-07:
Zwift forums seem to confirm that: [zwift](https://forums.zwift.com/t/schwinn-ic8-spin-bike/171514/295)

richlv, 2021-12-08:
As it seems to be connecting now, that might be it for this issue.
Thank you for the great help :)

dvmarinoff, 2021-12-08:
Cool, I am glad it now works for you now!
I am also taking this as an opportunity to refactor the whole bluetooth FTMS section of the code, something I wanted to do for awhile , but since it was working for everyone I was kinda hesitant to start changing it. Will merge into master in a few days and notify you here.

Cheers

closed 2021-12-10 by dvmarinoff

richlv, 2021-12-10:
Thank you, that sounds wonderful :)
BTW, Schwinn is not listed in https://github.com/dvmarinoff/Flux - could it be handled here, or would a new issue be better?

dvmarinoff, 2021-12-11:
Thanks for the reminder, just added it to the readme description.