dvmarinoff/Auuki #68: Very high values in the workout diagram overlap with other items or are cut of

991jo, 2021-12-10:
If the power of a workout is to high the diagram overlaps with the item above the list. This starts at ~150% FTP. See the attached image:

[image]

(Note the description is wrong, the test goes to 300% FTP).

A similar Issue exists on the home screen. Here the diagram is cut off

[image]

The same issue also exists for the power graph, but I would have to get on the trainer to get that picture.

dvmarinoff, 2021-12-10:
Now, that’s what they call an edge case :)
In general I haven’t though yet about support for FTP tests, they would need to be some kind of dynamic/generative workouts.

The specifics here are two:

- I am using little bit of css hackery with flexbox + regular html instead of proper graphing solution, and for now I’d prefer for things to stay that way.
- I will be changing all the code around the views soon, maybe a more general solution could be delayed for after that, just no point investing too much effort into this right now.

But until then maybe some kind of temporary solution would do fine.
I added a Ramp Test example and a very quick temp fix on the new zwo branch. It’s deployed here: [flux-zwo.vercel.app]( https://flux-zwo.vercel.app/) .

It looks like that for me now:

[image]

[image]

991jo, 2021-12-10:
Unless you are short, sprint-like efforts or a ramp test you probably wont run into that issue.

I can definitively understand your desire to keep the amount of dependencies small.
As a quick fix your current solution is fine.
I think you can close this issue or keep it as a reminder for when you want to overhaul the views. Your call.