dvmarinoff/Auuki #71: Setting a resistance target over bluetooth FTMS

dvmarinoff, 2021-12-11:
The bluetooth FTMS spec says use UINT8, the nRF Connect Simulator validates this approach, by accepting values in the range or 0 to 25.5 with a resolution of 0.1. But the some trainers disagree. The Tacx Flux returns invalid parameter when resistance target is set as UINT8 and the whole message is 2 bytes long as per spec, but works when it is set as UINT16 in the range of 0 to 100 with a resolution of 0.1.

Bluetooth FTMS:
[image]

The corresponding definition in the ANT+ FE-C:
[image]

Log from nRF Connect Simulator session with UINT8: [nrf-resistance-uint8.txt](https://github.com/dvmarinoff/Flux/files/7697467/nrf-resistance-uint8.txt)
Which shows that the Trainer simulator accepts the 2 byte long message, with values from 0 to 255 (0xFF).

```
[Server] WRITE REQUEST request for characteristic 00002ad9-0000-1000-8000-00805f9b34fb received, value: (0x) 04-F0
[Server] "Set Target Resistance Level Target Resistance Level: 24.0" received
server.sendResponse(GATT_SUCCESS, offset=0, value=null)
[Server] WRITE REQUEST request for characteristic 00002ad9-0000-1000-8000-00805f9b34fb received, value: (0x) 04-FA
[Server] "Set Target Resistance Level Target Resistance Level: 25.0" received
server.sendResponse(GATT_SUCCESS, offset=0, value=null)
[Server] WRITE REQUEST request for characteristic 00002ad9-0000-1000-8000-00805f9b34fb received, value: (0x) 04-04
[Server] "Set Target Resistance Level Target Resistance Level: 0.4" received
server.sendResponse(GATT_SUCCESS, offset=0, value=null)
```

Log from nRF Connect Simulator session with UINT16: [nrf-resistance-log-uint16.txt](https://github.com/dvmarinoff/Flux/files/7697465/nrf-resistance-log-uint16.txt)
Here the Trainer simulator accepts the 3 byte long message, but appears to reads only the second byte as per spec.

```
[Server] WRITE REQUEST request for characteristic 00002ad9-0000-1000-8000-00805f9b34fb received, value: (0x) 04-FA-00
[Server] "Set Target Resistance Level Target Resistance Level: 25.0" received
server.sendResponse(GATT_SUCCESS, offset=0, value=null)
[Server] WRITE REQUEST request for characteristic 00002ad9-0000-1000-8000-00805f9b34fb received, value: (0x) 04-04-01
[Server] "Set Target Resistance Level Target Resistance Level: 0.4" received
server.sendResponse(GATT_SUCCESS, offset=0, value=null)
[Server] WRITE REQUEST request for characteristic 00002ad9-0000-1000-8000-00805f9b34fb received, value: (0x) 04-0E-01
[Server] "Set Target Resistance Level Target Resistance Level: 1.4" received
server.sendResponse(GATT_SUCCESS, offset=0, value=null)
```

Log from the app itself against Tacx Flux S Trainer with Uint8, shows that the 2 byte message is considered invalid by the trainer implementation:

```
Fitness Machine Information:  {manufacturer: 'Tacx', model: 'T2900', firmware: '3.3.40'}
 :rx :ftms :response 'request control - success'
 :tx :ftms :resistance 1 -> 10
 :rx :ftms :response 'set target resistance - invalid parameter'
 :tx :ftms :resistance 2 -> 20
 :rx :ftms :response 'set target resistance - invalid parameter'
 :tx :ftms :resistance 3 -> 30
 :rx :ftms :response 'set target resistance - invalid parameter'
 :tx :ftms :resistance 4 -> 40
 :rx :ftms :response 'set target resistance - invalid parameter'
```

Here is another log this time with UINT16, which shows that the Tacx trainer expects a 3 byte long message and understands resistance target as a value between 0 and 1000: [flux-resistance-log-uint16.txt](https://github.com/dvmarinoff/Flux/files/7697525/flux-resistance-log-uint16.txt)

```
Fitness Machine Information:  {manufacturer: 'Tacx', model: 'T2900', firmware: '3.3.40'}
 :rx :ftms :response 'request control - success'
 :tx :ftms :resistance 24 -> 240
 :rx :ftms :status 'Target Resistance Level Changed' {"resistance":24}
 :rx :ftms :response 'set target resistance - success'
 :tx :ftms :resistance 25 -> 250
 :rx :ftms :status 'Target Resistance Level Changed' {"resistance":25}
 :rx :ftms :response 'set target resistance - success'
 :tx :ftms :resistance 26 -> 260
 :rx :ftms :status 'Target Resistance Level Changed' {"resistance":26}
 :rx :ftms :response 'set target resistance - success'
 :tx :ftms :resistance 27 -> 270
 :rx :ftms :status 'Target Resistance Level Changed' {"resistance":27}
 :rx :ftms :response 'set target resistance - success'
 :tx :ftms :resistance 100 -> 1000
 :rx :ftms :status 'Target Resistance Level Changed' {"resistance":100}
 :rx :ftms :response 'set target resistance - success'
 :tx :ftms :resistance 200 -> 2000
 :rx :ftms :response 'set target resistance - invalid parameter'
 :tx :ftms :resistance 101 -> 1010
 :rx :ftms :response 'set target resistance - invalid parameter'
 :tx :ftms :resistance 100 -> 1000
 :rx :ftms :status 'Target Resistance Level Changed' {"resistance":100}
 :rx :ftms :response 'set target resistance - success'
```

It reads values higher than 255 but lower than 1000.

I am not sure which is the best course of action, but my proposition is to send a 3 byte long message with target resistance as UINT16, since it doesn't seem to bother the simulator, and keeps Tacx happy. And also to lift the max allowed value up to 100 (1000). 

However 2 problems remain:
- an increment of 1 is too little and annoying for the 0 to 100 range, and an increment of 10 is too big for the 0 to 25.5 range.
- I can’t confirm that every trainer out there would tolerate the 3 byte long message and won’t treat it as an invalid parameter.

ihilt, 2021-12-12:
Looking through the [errata](https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=436247) (section 4.1) I think the issue might be that this was changed from `UINT8` to `SINT16`.  So, if I'm thinking about this correctly, that would be 2 bytes with a range of `-3,276.8` to `3,276.7` if it's using two's complement.  Strange that they made it a signed data type.

dvmarinoff, 2021-12-12:
[image]

Good job finding those, I was puzzled by the resistance target behaviour, but never thought to look in there.
The previous implementation had it that way, but I was afraid it was not spec compliant., it turns out it is.
Thanks for the help resolving that.

I think the singed type might be for trainers like the Tacx Neo that can simulate negative grade, they can probably do negative resistance too, if that makes sense. Although the the value range is plenty big and they can use an offset or something, but this way it is more intuitive.

closed 2021-12-12 by dvmarinoff