dvmarinoff/Auuki #76: run on local host

loopingleo, 2021-12-14:
Hi @dvmarinoff -
May I kindly ask you to show me how I can run the app on local host so I can test various ERG mode settings.
For instance, I would like to test whether I can keep my heart rate stable through adjusting the ERG power.
Thanks for your help and your great app!
Best,
Leo

dvmarinoff, 2021-12-14:
Thanks I am glad you find the app useful!

I personally use light-server for the development of this app, but there are many other options on npm that will do.

```
npm install light-server

light-server -s ./src/ -p 1234

 ```
Then open http://localhost:1234/ in a compatible browser from the table in the Readme.

loopingleo, 2021-12-15:
@dvmarinoff - do you have an I idea where in your code I could place a piece of code which automatically adjusts the target power so that heart rate stays stable at a predefined target heart rate?
Thanks. ✌🏼

dvmarinoff, 2021-12-15:
Now, I've gone a little nuts on the event-driven approach, I know, and currently I am in the process of putting some order into it. But the benefit is that experimental features can be done quickly.

If I correctly understand this is some kind of heart rate mode.

You can start by creating a file like [this](https://gist.github.com/dvmarinoff/465794fe4744b39971182e9f6cdc0f81) in the `/src/heart-rate-mode.js` folder. And then it needs to be included either at the bottom of the imports in `/src/index.js` for starts. I've given a basic version in the gist to get you started.

The events work that way:

if you need some data for example power, you need to sub for it like this:
```
xf.sub('db:power', function(power) {})

xf.sub('db:<some-db-property-name>',  function (<the-current-value-of-this-property>){})
```

If you want to send event or data you can do it with this:
```
xf.dispatch('db:power', 300)
xf.dispatch('db:heartRate', 140)
xf.dispatch('ui:power-target-set', 300)
xf.dispatch('ui:startWatch');
xf.dispatch('ui:startWorkout');
```

loopingleo, 2021-12-15:
Thanks, @dvmarinoff - this works really well. I have reduced the update check to every five seconds. 

Do you have an idea how to update the target heart rate value during the ride? Right now I have the target HR hard-coded in the file. But what if I want to update that value during the ride? At the moment I have to refresh the page and re-pair sensors for that.

dvmarinoff, 2021-12-15:
You, would need to create a UI for this. It's easy to extend, mostly copy paste and rename stuff, but you have to know what the general pattern is. Here is gist with the pieces of code that need to be added to each file: [gist](https://gist.github.com/dvmarinoff/e9523ed606c29e44e73a10a890533614)
You will need to figure out the line numbers.

This will create an additional tab along the mode controls, and will function exactly the same as the power target and the rest. Also the model and the prop inside db.

You can sub to the value with `xf.sub('db:heartRateTarget', (value) => {})`, inside your code.

loopingleo, 2021-12-15:
Wow, @dvmarinoff 😊👍🏻
Thanks a lot for your gists. 
With a few tweaks here and there I was able to create that new mode. 
First test ride went pretty well. 
I need to refine the power setting a bit to account for the lag in heart rate changes. 
I will let you know how when I am happy with the logic. 

I could image this could be used for solid 20 min FTP test pacing where you need to ensure your heart rate does not go way beyond threshold heart rate. 

Or, you could also use it for VO2max intervals where you want your heart rate to be just above threshold heart as soon as possible. I guess this is what I will try next. 

Ride On! 🤙🏿

dvmarinoff, 2021-12-16:
Thats’s true, heart rate is gonna lag behind, won’t be ease coming up with the algorithm. But I think this feature could also be useful for base training, where heart rate slowly drifts up, and the algorithm could adjust the power target inside the bounds of the targeted power zone.

Ride On!

closed 2022-01-12 by dvmarinoff