dvmarinoff/Auuki #90: Tacx Neo: gatt operation already in progress error

dvmarinoff, 2022-01-12:
Pedaling the bike while establishing a connection to a Tacx Neo trainer results in GATT server error messages on each control command, oddly enough the commands execution is successful. Pedaling the bike only after connection finishes does not result in error messages.

[image]

mjunkmyjunk, 2022-01-13:
Just curious why this is a problem? Seems to me on that this is just messages that says a connection is already the process of establishing.

dvmarinoff, 2022-01-13:
Failing subscription cleanup on disconnect wound explain the double `Disconnected` message at the end. Possibly the  multiple active subscribers, are sending multiple times the same command at the same time, hence: `GATT operation in progress`. But a page refresh would clean all subscribers, and still only the FEC BLE service fails with the same error even on 1 subscriber being active. While the first is a general BLE error on my side, the second I find puzzling.

I started integrating the new WebANT lib into the code base, will look to resolve this during the process.

dvmarinoff, 2022-01-16:
Wrong method was initiating the subs in controllable, which led to multiple handlers being created on reconnection.

closed 2022-01-16 by dvmarinoff