dvmarinoff/Auuki #92: add Power Match

dvmarinoff, 2022-01-13:
Match the controllable ERG target to the power meter output. Possibly by using a dynamic offset.

Notes:
- Trainers also have sweet spot power range when it comes to accuracy
- Flywheel speed,
- Temperature drift will mean that what is true for offset at minute 10 is not true at minute 40.
- Response time for HIIT style workouts like Chilli Pepper

TClin76, 2022-01-14:
I always have a question for Power Match in TrainerRoad.
Is it still in ERG mode for a smart trainer if APP use the power data from other pairing power meter? 
In this case, I think APP send resistance command to trainer, rather than target power. Is that correct?

mjunkmyjunk, 2022-01-14:
> In this case, I think APP send resistance command to trainer, rather than target power. Is that correct?

The app can send 
1) target power (using a fixed or dynamic offset to the Pedal Power Meter for eg)
2) A combo of Slope / Wind / Draft and such called "simulation mode" (ala zwift)

TClin76, 2022-02-09:
@dvmarinoff This is something related to the topic of power match, so I write in this thread.
Currently when trainer is pairing to FLUX, APP will occupy both power meter and controllable.

Is it possible to separate the pairing procedure for controllable and power meter? Just like what the most APP does.
In some cases, I would like to pair power meter and controllable to different APP. (like ROUVY and FLUX in the same time) 
Thanks

mjunkmyjunk, 2022-02-09:
if you're connecting via bluetooth, I don't think this is possible.
Or are you on ANT+?

TClin76, 2022-02-09:
@mjunkmyjunk 
OK ! I see
I did it on ANT+ before, but haven’t tried on BLUETOOTH. 
Thank u for prompt comment

MedTechCD, 2022-02-09:
Smart trainers can be paired in two ways: as a Controllable or as a Power source. Power meters obviously only as a Power source. Most Controllables and Power meters have only one BLE channel thus in BLE you can only pair one app to each. If they also have ANT+, you can pair any number of apps/devices in ANT+.
Just make sure that only one app/device is controlling the Smart Trainer! For the rest there are lots of possibilities.
I have used flux paired to the controllable and the power meter in BLE while my power meter was connected to my Garmin in ANT+. This way you can see on flux what the difference is between both.

dvmarinoff, 2022-02-09:
@TClin76 just added a basic tutorial on [How-To](https://github.com/dvmarinoff/Flux/discussions/101) use Flux with a second app. Will keep adding to it in the future.

TClin76, 2022-02-09:
Thank you @dvmarinoff, that is a great tutorial.

I do use ANT+ as the way for connecting Rouvy while using Flux via Bluetooth.
However ANT+ connection is not as stable as Bluetooth, I need to get another extended cable for ANT+ dongle in order to shorten the distance. So I start thinking if it is possible to connect both Apps via Bluetooth in the same time.
Now I know it has to be capable of the dual channel Bluetooth Trainer. 😂😂

mjunkmyjunk, 2022-02-15:
https://groups.google.com/g/golden-cheetah-users/c/vpUq4jn587Q/m/6sHmDPEKAgAJ

some discussion there.

z0nt, 2024-11-07:
I [implemented](https://github.com/z0nt/Auuki/tree/power-matcher) simple PowerMatch functionality which is watching for the average recent power and average power on the interval and makes adjustments up or down depending if averages are too high or too low.  It has 4 self explanatory settings which you can change to you taste.  I'm pretty happy with the results.  The average power is +/- 1 watt on 5 minutes steady state intervals compare to the target.  I like that the load slightly changes over time which simulates outdoor experience when you constantly need to change gears which requires small power and cadence changes as well.

Let me know if you want to try it yourself or you have any questions.

dvmarinoff, 2024-11-08:
@z0nt nice, currently I am on the road so I don't have a trainer or a power meter with me, but will take a look when I am back.