dvmarinoff/Auuki #96: Dual recording

dvmarinoff, 2022-01-19:
Record the data of two connected sources of power.

Options:
1. In a single .fit file with a custom developer field for the second power source
2. In two .fit files one for each source
3*. Why stop at 2 sensors, maybe someone would like to record at ones a whole set of power meters + a trainer. That could be up to 5 different sources of power.

TClin76, 2022-01-20:
Does .fit file natively support multiple power source records?

TClin76, 2022-01-21:
Understand. Thanks for the information.

mjunkmyjunk, 2022-02-15:
> 1. In a single .fit file with a custom developer field for the second power source

This is a good idea. But it will depend on the WEbSite if they're able to actually parse and display them. RunAnalyze? intervals.icu? Garmin Connect?

Interesting nevertheless