dvmarinoff/Auuki #97: Free Ride (switch between ERG and Slope)

TClin76, 2022-01-20:
I found there is an issue when .zwo file contain FreeRide command.
When Flux first time enters the FreeRide interval, it will switch rom ERG to Slope.
However when it leaves the FreeRide interval, it doesn't jump back to ERG mode.

https://user-images.githubusercontent.com/96434118/150307367-d5471426-4152-4f8e-9ed7-0c66b9cfbc7d.mp4

dvmarinoff, 2022-01-20:
See the app must have the ability to set the current mode, but also the User must have the ability to set the current mode and potentially  overwrite what the app/workout wants to do. It's a conflict of intent that I haven't decided yet on how to resolve, and just left Slope mode to always overwrite the other modes. 

One idea is to add a lock button that allows the User the ability to  lock the current mode and overwrite the app/workout, else if the lock button is off the app will take control back at the start of the next interval.

The problem: It's not super elegant solution and I couldn't easily find a place where to put the lock button inside the current UI so that it's easily clickable on mobile devices and not creating too much clutter.

One more thing is that I want to expand the options for Slope mode by adding UI that allows the User to pick values for
wind speed, coefficient of rolling resistance (tires) and wind resistance (how aero is the riders position).

I am open for ideas and suggestions.

TClin76, 2022-01-20:
I would suggest follow the way how Zwift
 Implement in workout.
<Unless user manually clicks “ERG OFF” to leave from erg to sim mode, erg always has higher priority than slope mode.>

If Flux switch from erg to slope due to FREERIDE command in .zwo, it should automatically switch back to erg when FREERIDE duration time is reached.
If Flux is set manually from erg to slope by user, it will stay in slope mode until user set it back to erg.

TClin76, 2022-01-25:
Just find another issue for .zwo parser.
It seems that Flux doesn't recognize "Ramp Duration" expression, but only "Warmup Duration" and "Cooldown Duration".

ex: <Ramp Duration="300" PowerLow="0.6" PowerHigh="0.7"/>

dvmarinoff, 2022-01-25:
Thanks, for finding those errors, you are really helping the project. I am currently fully focusing on finishing ANT+ support, which is a massive feature to bring to the browser. After I wrap things up with ANT+ I'll look into those.

dvmarinoff, 2022-01-30:
HI, just added support for Ramp .zwo element and some code that enforces the current interval mode, which fixes the FreeRide element behavior. It's on development version 0.1.26

closed 2022-01-31 by dvmarinoff