dvmarinoff/Auuki #99: Workout power gets "stuck"

GCuser99, 2022-02-02:
Hi @dvmarinoff,

Here are the results of a short test I ran today...

I created the following workout file for the test:
```
<workout_file>
  <author/>
  <name/>
  <description/>
  <sportType>bike</sportType>
  <durationType>time</durationType>
  <tags/>
  <workout>
    <Warmup Duration="295" PowerLow="0.054" PowerHigh="0.554" pace="0"/>
    <SteadyState Duration="120" Power="0.901" pace="0"/>
    <SteadyState Duration="120" Power="0.5" pace="0"/>
    <SteadyState Duration="120" Power="0.901" pace="0"/>
    <SteadyState Duration="120" Power="0.5" pace="0"/>
    <SteadyState Duration="120" Power="0.901" pace="0"/>
    <SteadyState Duration="120" Power="0.5" pace="0"/>
    <SteadyState Duration="120" Power="0.901" pace="0"/>
    <FreeRide Duration="240" FlatRoad="0"/>
  </workout>
</workout_file>
```
Next I succesfully paired my Wahoo Kickr and cadence sensors ...

I started the 23 minute workout, and after about 10 minutes the activities power profile in the display stopped updating and (I perceived) my output power remained constant at 180 watts throughout the remainder of the workout. Here is a snapshot that I took at the end of the workout:

[image]

What I noticed during the workout:

- After the power profile display stopped updating @+-10 minutes, power output remained stuck at 180 (actual pedaling resistance for a given cadence felt like it too), and average power stayed frozen at 177. All of the other indicators seemed to work fine, including cadence and target power. 
- In the workout, the last interval was free ride, and the display switched from Erg to Slope as expected, even though it felt like it was still stuck in Erg mode (180 watts).

I took a screen shot of the settings tab at the end of the ride and the Kickr no longer appears to be paired:

[image]

However, after uploading the fit file to Strava, it appears the entire 23 minute activity was captured and appears to verify that my power output was indeed 180 watts after the +-10 minute mark:

[image]

Here is the Developer Console output. Near the backing up of 600 records, the Ble for the controllable disconnected, but the cadence sensor remained connected until the end:
```
db.js:241 start db
views.js:12 start views.
connectionManager.js:9 start connection manager
index.js:22 start app.
idb.js:27 :idb :open :db 'store' :store-name 'session' ...
web-serial.js:214 
        
       :serial :ant-not-found
getKnownAnt @ web-serial.js:214
idb.js:93 :idb :getAll :store 'session' :success
models.js:458 :idb :restore 'session' :length 1
idb.js:93 :idb :clear :store 'session' :success
service.js:78 :rx :characteristics  Object
web-ble.js:151 Notifications started on cyclingPowerMeasurement: 00002a63-0000-1000-8000-00805f9b34fb.
web-ble.js:151 Notifications started on wahooTrainer: a026e005-0a7d-4ab3-97fa-f1500f9feb8b.
control.js:460 :rx :wcps :response [object Object] :raw 1,32,2
control.js:460 :rx :wcps :response [object Object] :raw 1,72,1,0,0,0
control.js:460 :rx :wcps :response [object Object] :raw 1,67,1,0
control.js:460 :rx :wcps :response [object Object] :raw 1,71,1,0,0,0
db.js:81 Object
service.js:78 :rx :characteristics  Object
common.js:84 maxRateCount: 3
web-ble.js:151 Notifications started on speedCadenceMeasurement: 00002a5b-0000-1000-8000-00805f9b34fb.
[... 110 lines cut ...]
watch.js:277 backing up of 1080 records ...
idb.js:93 :idb :put :store 'session' :success
db.js:122 :zwo set target cadence 0
watch.js:277 backing up of 1140 records ...
idb.js:93 :idb :put :store 'session' :success
watch.js:277 backing up of 1200 records ...
idb.js:93 :idb :put :store 'session' :success
watch.js:277 backing up of 1260 records ...
idb.js:93 :idb :put :store 'session' :success
watch.js:277 backing up of 1320 records ...
idb.js:93 :idb :put :store 'session' :success
watch.js:53 Workout done!
watch.js:38 Workout done!
watch.js:277 backing up of 1380 records ...
idb.js:93 :idb :put :store 'session' :success
db.js:122 :zwo set target cadence 0
watch.js:277 backing up of 1400 records ...
idb.js:93 :idb :put :store 'session' :success
models.js:394 Array(1422)
lock.js:51 Wake lock released.
idb.js:93 :idb :put :store 'session' :success
lock.js:51 Wake lock released.
idb.js:93 :idb :put :store 'session' :success
device.js:87 Disconnected ble:speed-and-cadence, Wahoo CADENCE DF21.
```

dvmarinoff, 2022-02-03:
Thanks for the report, it’s incredibly valuable. And I am sorry to hear your workout didn't went through.

- The protocol
Looking at the log, I see the trainer is using the wahoo cycling power service (wcps). That’s an old Wahoo specific proprietary protocol from the days before the now standard bluetooth FTMS, and the bad news about it is that it’s also undocumented. Officially it is not supported by this project, but unofficially I have tried to do my best in hunting down and stitching together various bits and pieces from partial implementations around the internet. I couldn’t find anyone with a trainer that uses this protocol and is willing to help test it, so you won’t see it listed as officially supported. My implementation remained untested on real device and I am amazed it’s even working. So your log here is the first and only feedback I have.  

- The disconnect
I am not sure what triggered the drop out. It might be a coincidence or something protocol related, but auto-reconnects over web-bluetooth are a special case and are not supported at this point (they require a specific option in the browser settings to be checked out).
If the connection drops, you would have to manually connect back to the device. Did you try reconnecting to the trainer?

In my case, my Tacx Flux S trainer has a well known firmware bug, which makes it drop the connection when you stop pedalling after a prolong steady effort at intensity. But in any case I get drop outs rarely, like 1 a week, and it’s quick and easy to reconnect the controllable from the top menu.

The rest is more or less expected behaviour. The workout kept running, and the power value remained at the last data point it received. What I can do is reset values to 0 when the connection is dropped, but trying to distinguish between real signal and signal that’s stuck is hard in this case (some trainers seriously over smooth data when in ERG, etc., and such a feature is likely to cause more problems than solve any).

GCuser99, 2022-02-03:
I'm glad the test was helpful feedback. It is pretty amazing that your code works as well as it does without testing. 

I didn't try to reconnect the device because I really didn't understand what was going on until after the workout was completed. I will try again to see if the dropout was just a fluke.

As for a fix other than an auto-reconnect, which you're not supporting now, I wouldn't bother... Maybe if you can get the Ant+ side of this working, in Windows someday... :-)

dvmarinoff, 2022-02-03:
Hm, I'll have to think of some other way to communicate disconnections. Maybe a text message could show up on top of the current power graph. Else the red-yellow-green indicator on the side of the controllable label, at the top bar indicates the current status of the connection: disconnected-connecting-connected. But I guess it's confusing when the resistance remains the same and the clock is ticking. My trainer resets to 0 resistance on disconnection, and the sudden drop in resistance is quite clear sign of what is going on.

GCuser99, 2022-02-03:
I know now to look at the red-yellow-green indicator light. I wasn't really confused and knew something was wrong, but wanted to finish the test without complicating things. And besides I needed the exercise! Thanks again for sharing your software...

TClin76, 2022-02-08:
@dvmarinoff BLE unexpected disconnection also happened in my case, but not so often.
However my trainer (Magene) handle controllable disconnection is quite different from your TACX Flux.
It seems that the ERG is still working, therefore I couldn't notice from resistance suddenly dropped.
I think it is a good way to reset power value to 0 which is much easier for user to notice that the connection is dropped.

BTW, I can manually re-connect the device without any problem.

dvmarinoff, 2022-02-08:
@TClin76 Thanks for the report!

If I am not mistaken thats on MacOS. Can you share more about your setup and also how often you get the drop outs?

dvmarinoff, 2022-02-09:
I am working on bluetooth auto-reconnects, but will have to withhold them for a few more days, I am having some issues with the scanning API. A few more stability improvements are coming along too, but need more testing.

dvmarinoff, 2022-02-16:
Closing this one. Will track progress of web-bluetooth auto reconnects [here](https://github.com/dvmarinoff/Flux/issues/106).

Did a lot of work on the issue last week, but with little progress. Some unexpected web-bluetooth behavior turned out to be related to a current bug in MacOS bluetooth which was only partially resolved by the latest update. Will do some easy features this week and resume work on bluetooth in the next one.

closed 2022-02-16 by dvmarinoff