dvmarinoff/Auuki #100: Is Slope Mode Working in WCPS?

GCuser99, 2022-02-03:
Hi @dvmarinoff,

I tried a test today to see how slope mode works. I'll admit up-front that I may have misunderstood Slope targets from Issue #26, but here is what I tried on the latest Dev App with Chrome browser:

```
<workout_file>
  <author/>
  <name/>
  <description/>
  <sportType>bike</sportType>
  <durationType>time</durationType>
  <tags/>
  <workout>
    <Warmup Duration="60" PowerLow="0.054" PowerHigh="0.554" pace="80"/>
    <SteadyState Duration="60" SlopeTarget="0.0" pace="80"/>
    <SteadyState Duration="60" SlopeTarget="15.0" pace="80"/>
    <SteadyState Duration="60" SlopeTarget="0.0" pace="80"/>	
    <SteadyState Duration="60" SlopeTarget="15.0" pace="80"/>
    <SteadyState Duration="60" SlopeTarget="0.0" pace="80"/>
  </workout>
</workout_file>
```
I expected that after the warmup I would get 60 secs of flat slope free ride, followed by 15% slope free ride, etc...

What I observed on the display was that the workout never switched out of Erg mode and very little resistance after the warmup.

BTW, I lost Kickr bluetooth connection after about 3 minutes, but not before the first 15% slope interval was nearly completed.

Here is the console log:
```
db.js:241 start db
views.js:12 start views.
connectionManager.js:9 start connection manager
index.js:22 start app.
idb.js:27 :idb :open :db 'store' :store-name 'session' ...
idb.js:93 :idb :getAll :store 'session' :success
models.js:458 :idb :restore 'session' :length 1
idb.js:93 :idb :clear :store 'session' :success
web-serial.js:214 :serial :ant-not-found
getKnownAnt @ web-serial.js:214
service.js:78 :rx :characteristics  Object
web-ble.js:151 Notifications started on cyclingPowerMeasurement: 00002a63-0000-1000-8000-00805f9b34fb.
web-ble.js:151 Notifications started on wahooTrainer: a026e005-0a7d-4ab3-97fa-f1500f9feb8b.
control.js:460 :rx :wcps :response [object Object] :raw 1,32,2
control.js:460 :rx :wcps :response [object Object] :raw 1,72,1,0,0,0
control.js:460 :rx :wcps :response [object Object] :raw 1,67,1,0
control.js:460 :rx :wcps :response [object Object] :raw 1,71,1,0,0,0
common.js:262 rateAdjuster :on pm :tsAvgDiff 1004.0500000000001 :result 2
common.js:84 maxRateCount: 2
common.js:84 maxRateCount: 2
db.js:81 Object
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,10,0
service.js:78 :rx :characteristics  Object
common.js:84 maxRateCount: 3
web-ble.js:151 Notifications started on speedCadenceMeasurement: 00002a5b-0000-1000-8000-00805f9b34fb.
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,30,0
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,50,0
common.js:262 rateAdjuster :on cscs :tsAvgDiff 821.6999999999998 :result 3
common.js:84 maxRateCount: 3
common.js:84 maxRateCount: 3
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,70,0
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,90,0
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,110,0
watch.js:277 backing up of 60 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,0,0
watch.js:277 backing up of 120 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,0,0
device.js:87 Disconnected ble:controllable, Wahoo KICKR A730.
watch.js:277 backing up of 180 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
db.js:122 :zwo set target cadence 0
watch.js:277 backing up of 195 records ...
idb.js:93 :idb :put :store 'session' :success
```

GCuser99, 2022-02-04:
I just noticed that "SlopeTarget" may not be the operative keyword, so I tried "Slope" instead...
```
<workout_file>
  <author/>
  <name/>
  <description/>
  <sportType>bike</sportType>
  <durationType>time</durationType>
  <tags/>
  <workout>
    <Warmup Duration="60" PowerLow="0.054" PowerHigh="0.554" pace="80"/>
    <SteadyState Duration="60" Slope="0.0" pace="80"/>
    <SteadyState Duration="60" Slope="20.0" pace="80"/>
    <SteadyState Duration="60" Slope="0.0" pace="80"/>	
    <SteadyState Duration="60" Slope="20.0" pace="80"/>
    <SteadyState Duration="60" Slope="0.0" pace="80"/>
  </workout>
</workout_file>
```
What I noticed this time is that after the warmup, the mode did switch from Erg to Slope, and the slope indicator correctly updated for each of the slope intervals. However, the power remained at 110 regardless of my cadence, and I felt no difference between 0% and 20% slope. It felt like I was in 110 watts Erg mode after the warmup.

Here is the console log:
```
db.js:241 start db
views.js:12 start views.
connectionManager.js:9 start connection manager
index.js:22 start app.
idb.js:27 :idb :open :db 'store' :store-name 'session' ...
idb.js:93 :idb :getAll :store 'session' :success
models.js:458 :idb :restore 'session' :length 1
idb.js:93 :idb :clear :store 'session' :success
web-serial.js:214 :serial :ant-not-found
getKnownAnt @ web-serial.js:214
service.js:78 :rx :characteristics  Object
web-ble.js:151 Notifications started on cyclingPowerMeasurement: 00002a63-0000-1000-8000-00805f9b34fb.
web-ble.js:151 Notifications started on wahooTrainer: a026e005-0a7d-4ab3-97fa-f1500f9feb8b.
control.js:460 :rx :wcps :response [object Object] :raw 1,32,2
control.js:460 :rx :wcps :response [object Object] :raw 1,72,1,0,0,0
control.js:460 :rx :wcps :response [object Object] :raw 1,67,1,0
service.js:78 :rx :characteristics  Object
common.js:84 maxRateCount: 3
web-ble.js:151 Notifications started on speedCadenceMeasurement: 00002a5b-0000-1000-8000-00805f9b34fb.
control.js:460 :rx :wcps :response [object Object] :raw 1,71,1,0,0,0
db.js:81 Object
common.js:262 rateAdjuster :on pm :tsAvgDiff 1022.05 :result 2
common.js:84 maxRateCount: 2
common.js:84 maxRateCount: 2
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,10,0
common.js:262 rateAdjuster :on cscs :tsAvgDiff 1057.95 :result 2
common.js:84 maxRateCount: 2
common.js:84 maxRateCount: 2
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,30,0
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,50,0
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,70,0
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,90,0
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,66,1,0,110,0
watch.js:277 backing up of 60 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:460 :rx :wcps :response [object Object] :raw 1,70,64
watch.js:277 backing up of 120 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:460 :rx :wcps :response [object Object] :raw 1,70,64
watch.js:277 backing up of 180 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:460 :rx :wcps :response [object Object] :raw 1,70,64
watch.js:277 backing up of 240 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:460 :rx :wcps :response [object Object] :raw 1,70,64
device.js:87 Disconnected ble:controllable, Wahoo KICKR A730.
watch.js:277 backing up of 300 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
db.js:122 :zwo set target cadence 0
watch.js:277 backing up of 318 records ...
idb.js:93 :idb :put :store 'session' :success
lock.js:51 Wake lock released.
idb.js:93 :idb :put :store 'session' :success
```

GCuser99, 2022-02-04:
On the ble disconnects - I am wondering if my Zwift Steerzo might be interfering - I will try with that turned off to see if I can get through a workout w/o disconnect...

@dvmarinoff , if there are additional tests that you would like me to perform to help test/debug, just provide instructions and I'll give it a go!

GCuser99, 2022-02-04:
Also, I removed "pace" parameter (which I now see should be "Cadence") and tried again - the results were same as above. Felt like Erg mode @110 watts.

dvmarinoff, 2022-02-04:
### WSPC

The log shows that the last successful message is the ERG command to set power target at 110.
`1,66,1,0,110,0`, where 66 is the number code for ERG mode, 110 is the Watts and 1 possibly means success.
The Slope mode command received `1,70,64`, I don’t know the meaning of the error codes, but 64 isn’t good, should be 1.

Option 1: my guess about the format of the Slope command was wrong, it’s good that it got recognised as such at all, so probably the formatting of the slope target value itself is the error.

Option 2: I need to send SIM mode parameters command before the Slope command
That’s a wild guess, but in FTMS those 2 messages form 1 single message that carries all the needed parameters + the slope value. I do send SIM command at the start of the session which configures those parameters, so it shouldn’t be needed again unless they change.

I’ll check those, do some changes to the code and will ask you if can do another test run. This may take a couple if iterations until I get it right, but with the lack of documentation and no access to the hardware device I am flying blind here.

### Disconnects

I wouldn’t blame the sterzo, but you could try. My setup is like an interference stress test and I don’t get drop outs. It includes wireless earbuds, RGT or Zwift, Companion App and Flux all running together, sometimes all on bluetooth, sometimes ANT+ and bluetooth.

- **Option 1**: It could be malformed command on my side that crashes.
Not much evidence on this, because it disconnects in ERG mode where all commands pass with success.

- **Option 2**: the trainer is expecting a particular configuration and when it doesn’t receive it, it just shuts down after awhile.
The response code to the first configuration message (‘unlock’) is 2. I don’t know what it means, but it’s not what it should be.

- **Option 3**: It could be the bluetooth wireless module on your trainer or your laptop.
Do you get disconnects when using bluetooth with Zwift?

Option 4: Web Bluetooth is not reliable on Windows 10. I just did 2 full workouts Dijon and Pasta on my Thinkpad X240 with Windows 10, no dropouts. I think I can rule this out.

### Slope targets
You got that right I do use `Slope=“<number>”` for slope targets, you can use it alone or with a companion `Power=“<number>”`.

### Meanwhile
I started working on automatic re-connects over bluetooth, will get a working version of them by Monday.

I also did some attempts to get WebANT working on Windows 10. There is 1 good and 1 bad news.
The good one is that I got it working, but it is complicated. The Chrome browser WebUSB API requires different operating system driver to be loaded than the default one. If you switch the driver from libusb to winusb, it works I’ll be tracking this in more details in [here](https://github.com/dvmarinoff/WebANT/issues/2). The google team that works on WebUSB recommended using [Zadig](https://zadig.akeo.ie/) utility tool for switching the drivers.

The bad news is that this requires an additional step from Users. They will need to install utility tool and switch drivers. The worst is that Zwift which depends on the default driver will stop recognizing the usb stick until you switch the drivers back. That’s going to be very hard to communicate to Users and I don’t think that at this point I’ll have official support for WebANT on Windows 10. I’ll keep digging though maybe something will come up.

GCuser99, 2022-02-04:
Ok, I'm willing to do a set of iterative tests when you need me to. 
Here is one I thought of to address your Sim/Slope pairs issue:
**Sim/Slope Test:**
Chrome browser, Dev Vercel
This test is never asked to go into Erg mode to see if Erg is somehow interfering with Sim parameters:
```
<workout_file>
  <author/>
  <name/>
  <description/>
  <sportType>bike</sportType>
  <durationType>time</durationType>
  <tags/>
  <workout>
    <SteadyState Duration="120" Slope="20.0"/>
    <SteadyState Duration="60" Slope="0.0"/>
  </workout>
</workout_file>
```
It felt like a level free ride the entire 3 minutes:

```
db.js:241 start db
views.js:12 start views.
connectionManager.js:9 start connection manager
index.js:22 start app.
idb.js:27 :idb :open :db 'store' :store-name 'session' ...
idb.js:93 :idb :getAll :store 'session' :success
models.js:458 :idb :restore 'session' :length 1
idb.js:93 :idb :clear :store 'session' :success
web-serial.js:214 :serial :ant-not-found
getKnownAnt @ web-serial.js:214
service.js:78 :rx :characteristics  Object
web-ble.js:151 Notifications started on cyclingPowerMeasurement: 00002a63-0000-1000-8000-00805f9b34fb.
web-ble.js:151 Notifications started on wahooTrainer: a026e005-0a7d-4ab3-97fa-f1500f9feb8b.
db.js:81 Object
control.js:460 :rx :wcps :response [object Object] :raw 1,32,2
control.js:460 :rx :wcps :response [object Object] :raw 1,72,1,0,0,0
controllable.js:54 Uncaught TypeError: Cannot read properties of undefined (reading 'setTargetSlope')
    at Controllable.onSlopeTarget (/ble/controllable.js:54:26)
    at handlerWraper (/functions.js:273:21)
    at dispatch (/functions.js:263:16)
    at Object.set (/functions.js:255:17)
    at /db.js:136:20
    at /functions.js:286:49
    at Object.dispatch (/functions.js:263:16)
    at /db.js:100:12
    at /functions.js:286:49
    at Object.dispatch (/functions.js:263:16)
db.js:122 :zwo set target cadence 0
control.js:460 :rx :wcps :response [object Object] :raw 1,67,1,0
control.js:460 :rx :wcps :response [object Object] :raw 1,71,1,0,0,0
service.js:78 :rx :characteristics  Object
common.js:84 maxRateCount: 3
web-ble.js:151 Notifications started on speedCadenceMeasurement: 00002a5b-0000-1000-8000-00805f9b34fb.
common.js:262 rateAdjuster :on pm :tsAvgDiff 985.6000000000001 :result 2
common.js:84 maxRateCount: 2
common.js:84 maxRateCount: 2
common.js:262 rateAdjuster :on cscs :tsAvgDiff 1006.7000000000003 :result 2
common.js:84 maxRateCount: 2
common.js:84 maxRateCount: 2
watch.js:277 backing up of 60 records ...
idb.js:93 :idb :put :store 'session' :success
watch.js:277 backing up of 120 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:460 :rx :wcps :response [object Object] :raw 1,70,1,0,0,128
watch.js:277 backing up of 180 records ...
watch.js:53 Workout done!
watch.js:38 Workout done!
idb.js:93 :idb :put :store 'session' :success
db.js:122 :zwo set target cadence 0
watch.js:277 backing up of 187 records ...
idb.js:93 :idb :put :store 'session' :success
control.js:460 :rx :wcps :response [object Object] :raw 1,70,1,0,153,153
lock.js:51 Wake lock released.
idb.js:93 :idb :put :store 'session' :success
device.js:87 Disconnected ble:controllable, Wahoo KICKR A730.
```

dvmarinoff, 2022-02-04:
Excellent!

The order of events is:

`1,32,2`, initialise Wahoo characteristic, response code 2
`1,72,1,0,0,0`, setWheelCircumference, success, value 0 
`1,67,1,0`, setSimMode, success
`1,71,1,0,0,0`, setWindSpeed, success, value 0
`1,70,1,0,0,128`, setSlope, success, value 0
`1,70,1,0,153,153`, setSlope, success, value 20

The setSlope command passes with success. So it is ERG that’s braking the Slope target command.

But the odd thing is that the commands are executed in reverse order and shifted 2 minutes ahead, with the one setting grade to 20 happening after the Workout done message.

I pushed a new development version, you may need to refresh the page to get it. It has updated logs with timestamps and every Slope target command is preceeded with a setSIM command.

When are around it can you repeat a similar test, but also use the freeform +/- buttons to update the Slope Target and see if something happens.

GCuser99, 2022-02-04:
Ok, I finally made it through 2x10 minute workouts without dropping the controllable. The first workout I'm not including here because the cadence sensor dropped after the first several minutes (first time that has ever happened!). I will do some tests on a different windows pc later to rule in/out the Bluetooth stability of the computer. I will also do a test where I connect Zwift through Bluetooth to see if trainer stays connected to rule in/out trainer as the cuplrit for drops.

This test used Chrome and your very latest dev vercel app:

```
<workout_file>
  <author/>
  <name/>
  <description/>
  <sportType>bike</sportType>
  <durationType>time</durationType>
  <tags/>
  <workout>
    <SteadyState Duration="60" Slope="0.0"/>
    <SteadyState Duration="120" Slope="10.0"/>
    <SteadyState Duration="60" Slope="15.0"/>
    <SteadyState Duration="120" Slope="0.0"/>
    <SteadyState Duration="60" Power="0.75"/>
    <SteadyState Duration="60" Slope="10.0"/>
    <FreeRide Duration="120" FlatRoad="0"/>
  </workout>
</workout_file>
```
Summary: all of the non-zero slope intervals felt same as zero slope. The Erg interval worked fine, and then switched back into Slope mode as expected. At around the 5 minute and 9 minute marks I tried adjusting the slope manually to no effect. An observation that you made about the last test occurred in this one too - during the manual adjusts some the slope commands showed up in the log out of order. For easier interpretation, I inserted descriptive lines in the log at each of the workout interval boundaries...
```
db.js:241 start db
views.js:12 start views.
connectionManager.js:9 start connection manager
index.js:22 start app.
idb.js:27 :idb :open :db 'store' :store-name 'session' ...
web-serial.js:214 :serial :ant-not-found
getKnownAnt @ web-serial.js:214
idb.js:93 :idb :getAll :store 'session' :success
models.js:458 :idb :restore 'session' :length 1
idb.js:93 :idb :clear :store 'session' :success
service.js:78 :rx :characteristics  Object
web-ble.js:151 Notifications started on cyclingPowerMeasurement: 00002a63-0000-1000-8000-00805f9b34fb.
web-ble.js:151 Notifications started on wahooTrainer: a026e005-0a7d-4ab3-97fa-f1500f9feb8b.
control.js:475 1644009598154 :rx :wcps :response 1,32,2
control.js:245 1644009599155 :tx :wcps :circumference [object Object]
control.js:475 1644009599292 :rx :wcps :response 1,72,1,0,0,0
control.js:356 1644009600296 :tx :wcps :sim 75, 0.004, 0.51
control.js:475 1644009600854 :rx :wcps :response 1,67,1,0
control.js:304 1644009601625 :tx :wcps :windSpeed [object Object]
control.js:475 1644009601934 :rx :wcps :response 1,71,1,0,0,0
common.js:262 rateAdjuster :on pm :tsAvgDiff 1003.5499999999998 :result 2
common.js:84 maxRateCount: 2
common.js:84 maxRateCount: 2
service.js:78 :rx :characteristics  Object
common.js:84 maxRateCount: 3
web-ble.js:151 Notifications started on speedCadenceMeasurement: 00002a5b-0000-1000-8000-00805f9b34fb.
db.js:81 Object
******************** START OF INTERVAL 1 (60 secs @ Slope=0) ********************
wcps.js:69 slope
control.js:356 1644009628730 :tx :wcps :sim 75, 0.004, 0.51
db.js:122 :zwo set target cadence 0
control.js:475 1644009629595 :rx :wcps :response 1,67,1,0
control.js:194 1644009630596 :tx :wcps :slope 0
control.js:475 1644009630854 :rx :wcps :response 1,70,1,0,0,128
common.js:262 rateAdjuster :on cscs :tsAvgDiff 872.05 :result 2
common.js:84 maxRateCount: 2
common.js:84 maxRateCount: 2
watch.js:277 backing up of 60 records ...
******************** START OF INTERVAL 2 (120 secs @ Slope=10) ********************
wcps.js:69 slope
control.js:356 1644009688738 :tx :wcps :sim 75, 0.004, 0.51
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:475 1644009689355 :rx :wcps :response 1,67,1,0
control.js:194 1644009690184 :tx :wcps :slope 10
[... 463 lines cut ...]
    at Summary (/fit/activity.js:214:43)
    at Object.encode (/fit/activity.js:255:26)
    at Workout.encode (/models/models.js:393:40)
    at Workout.save (/models/models.js:404:28)
    at /db.js:178:24
    at /functions.js:286:49
    at Object.dispatch (/functions.js:263:16)
    at HTMLDivElement.onSave (/views/watch.js:44:21)
(anonymous) @ db.js:181
db.js:181 Error on activity save:  TypeError: Cannot read properties of undefined (reading 'timestamp')
    at Summary (/fit/activity.js:214:43)
    at Object.encode (/fit/activity.js:255:26)
    at Workout.encode (/models/models.js:393:40)
    at Workout.save (/models/models.js:404:28)
    at /db.js:178:24
    at /functions.js:286:49
    at Object.dispatch (/functions.js:263:16)
    at HTMLDivElement.onSave (/views/watch.js:44:21)
(anonymous) @ db.js:181
lock.js:51 Wake lock released.
idb.js:93 :idb :put :store 'session' :success
```
The resulting fit file loaded up in Strava fine. The two short zero cadence/power intervals correspond to when I reached over to my laptop to manually adjust the slope (approx 5 minutes and 9 minutes).

[image]

GCuser99, 2022-02-04:
BTW, I ran several additional tests by increasing the slopes by two orders of magnitude (10,15 to 100,150 to 1000,1500) to rule out a units problem. Unfortunately for these tests, it looks like the code limits slope to 40. However, I felt no difference even at 40.

GCuser99, 2022-02-05:
Another thought occurred to me similar to above - should/can you check sim parameters to make sure that weight is being encoded properly? For example, if weight is set to zero inadvertently, then slope should have no effect, right?

GCuser99, 2022-02-05:
Sorry for the many comments - but I just tried a test setting my weight to 500 kg, reasoning that even on a flat road, I should feel a difference in resistance from the default of 75 kg in sim mode. But the developer console showed wcps :sim 75, 0.004, 0.51. Then I set my weight to 70kg, and it still showed 75 in the console. Can you check to make sure the user input weight widget is working?

dvmarinoff, 2022-02-05:
## Weight
Yep, the User weight is transmitted over the custom controllable protocol and I’ve set it to a fixed value 75kg / 165lbs, for simplicity. I’ll hook up the weight widget once we figure out the controllable protocol.

Else I’am using GoldenCheetah and [SensorsSwiftTrainers](https://github.com/codeinversion/sensors-swift-trainers/blob/master/Sources/SwiftySensorsTrainers/WahooTrainerSerializer.swift) projects as main sources.

In my code I’ve developed a declarative approach for defining the constraints, scaling and formatting of values like `weight`, that works across protocols. On line 331, you can see that the resolution of weight is defined as 0.01. This means that the weight value 74 will be divided by 0.01, which produces the same result as multiplying by 100.
https://github.com/dvmarinoff/Flux/blob/51babc308a206c900141550a204a779745367081/src/ble/wcps/control.js#L325

This is my test file for WCPS. Line 303 checks that the weight 74 is written as 7400.
https://github.com/dvmarinoff/Flux/blob/cab5c20b1ba33245e889ff79ef68d15ac6d84b01/test/ble/wcps.test.js#L334

## Slope
Here is the Slope Target code. At line 160 and 161, I have for comparison, the encoding of Slope from SensorsSwiftTrainers and GoldenCheetah respectively. and my implementation is at line 178, it matches exactly the one from GC and when combined with the `definitions` spec it produces the same result as SST.

https://github.com/dvmarinoff/Flux/blob/51babc308a206c900141550a204a779745367081/src/ble/wcps/control.js#L154

The encoding should be right. It allows for values between -100 and +100 to be written as slope. If we remove one zero, the max possible value would be 10%.

Now I do remember that on wahoo trainers Wheel Circumference has effect on resistance in SIM mode. I’ll check this too and report back

dvmarinoff, 2022-02-05:
OK, checked and in fact I was sending wrongly formatted arguments to wheel circumference and wind speed which probably resulted in them being 0. Didn't have the time to write tests now, but push some code with fixes. Hopefully this will work. Will return later ro examine this further and write proper test cases.

GCuser99, 2022-02-05:
Good catch @dvmarinoff! Well the good news is that WE GOT SLOPE! In the following test, I really felt the slope changes. I'm going to do some more analysis of my ride to see if the power-speed numbers are reasonable. I'm not 100% sure, but it did feel like the speed might have been a low for my power output...
```
<workout_file>
  <author/>
  <name/>
  <description/>
  <sportType>bike</sportType>
  <durationType>time</durationType>
  <tags/>
  <workout>
    <SteadyState Duration="60" Slope="0.0"/>
    <SteadyState Duration="60" Slope="5.0"/>
    <SteadyState Duration="60" Slope="10.0"/>
    <SteadyState Duration="60" Slope="0.0"/>
  </workout>
</workout_file>
```
```
db.js:241 start db
views.js:12 start views.
connectionManager.js:9 start connection manager
index.js:22 start app.
idb.js:27 :idb :open :db 'store' :store-name 'session' ...
idb.js:93 :idb :getAll :store 'session' :success
models.js:458 :idb :restore 'session' :length 1
idb.js:93 :idb :clear :store 'session' :success
web-serial.js:214 :serial :ant-not-found
getKnownAnt @ web-serial.js:214
service.js:78 :rx :characteristics  Object
web-ble.js:151 Notifications started on cyclingPowerMeasurement: 00002a63-0000-1000-8000-00805f9b34fb.
web-ble.js:151 Notifications started on wahooTrainer: a026e005-0a7d-4ab3-97fa-f1500f9feb8b.
control.js:475 1644081967994 :rx :wcps :response 1,32,2
control.js:245 1644081968755 :tx :wcps :circumference 21050
control.js:475 1644081969074 :rx :wcps :response 1,72,1,0,58,82
control.js:356 1644081970086 :tx :wcps :sim 75, 0.004, 0.51
control.js:475 1644081970514 :rx :wcps :response 1,67,1,0
control.js:304 1644081971397 :tx :wcps :windSpeed 34768
control.js:475 1644081971894 :rx :wcps :response 1,71,1,0,208,135
service.js:78 :rx :characteristics  Object
common.js:84 maxRateCount: 3
web-ble.js:151 Notifications started on speedCadenceMeasurement: 00002a5b-0000-1000-8000-00805f9b34fb.
db.js:81 Object
common.js:262 rateAdjuster :on pm :tsAvgDiff 998.05 :result 2
common.js:84 maxRateCount: 2
common.js:84 maxRateCount: 2
common.js:262 rateAdjuster :on cscs :tsAvgDiff 866.0500000000001 :result 2
common.js:84 maxRateCount: 2
common.js:84 maxRateCount: 2
*************************** START SLOPE=0 ***************************
wcps.js:71 slope
control.js:356 1644082005494 :tx :wcps :sim 75, 0.004, 0.51
db.js:122 :zwo set target cadence 0
control.js:475 1644082006397 :rx :wcps :response 1,67,1,0
control.js:194 1644082007043 :tx :wcps :slope 0
control.js:475 1644082007477 :rx :wcps :response 1,70,1,0,0,128
watch.js:277 backing up of 60 records ...
*************************** START SLOPE=5 ***************************
wcps.js:71 slope
control.js:356 1644082065513 :tx :wcps :sim 75, 0.004, 0.51
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:475 1644082066337 :rx :wcps :response 1,67,1,0
control.js:194 1644082067159 :tx :wcps :slope 5
control.js:475 1644082068138 :rx :wcps :response 1,70,1,0,102,134
watch.js:277 backing up of 120 records ...
*************************** START SLOPE=10 **************************
wcps.js:71 slope
control.js:356 1644082125511 :tx :wcps :sim 75, 0.004, 0.51
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:475 1644082126098 :rx :wcps :response 1,67,1,0
control.js:194 1644082126930 :tx :wcps :slope 10
control.js:475 1644082127537 :rx :wcps :response 1,70,1,0,204,140
watch.js:277 backing up of 180 records ...
*************************** START SLOPE=0 ***************************
wcps.js:71 slope
control.js:356 1644082185514 :tx :wcps :sim 75, 0.004, 0.51
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:475 1644082186938 :rx :wcps :response 1,67,1,0
control.js:194 1644082187600 :tx :wcps :slope 0
control.js:475 1644082187838 :rx :wcps :response 1,70,1,0,0,128
watch.js:277 backing up of 240 records ...
*************************** WORKOUT DONE ***************************
watch.js:53 Workout done!
watch.js:38 Workout done!
idb.js:93 :idb :put :store 'session' :success
wcps.js:71 slope
control.js:356 1644082258268 :tx :wcps :sim 75, 0.004, 0.51
db.js:122 :zwo set target cadence 0
watch.js:277 backing up of 249 records ...
idb.js:93 :idb :put :store 'session' :success
control.js:475 1644082258937 :rx :wcps :response 1,67,1,0
control.js:194 1644082259580 :tx :wcps :slope 0
control.js:475 1644082260378 :rx :wcps :response 1,70,1,0,0,128
models.js:394 Array(266)
control.js:82 1644082260668 :tx :wcps :loadIntensity 0
wcps.js:71 slope
control.js:356 1644082260668 :tx :wcps :sim 75, 0.004, 0.51
web-ble.js:171 characteristic.writeValue: DOMException: GATT operation already in progress.
writeCharacteristic @ web-ble.js:171
control.js:475 1644082261638 :rx :wcps :response 1,64,1,0,255,63
control.js:194 1644082261683 :tx :wcps :slope 0
control.js:475 1644082262357 :rx :wcps :response 1,70,64
lock.js:51 Wake lock released.
idb.js:93 :idb :put :store 'session' :success
device.js:87 Disconnected ble:speed-and-cadence, Wahoo CADENCE DF21.
```
[image]

GCuser99, 2022-02-06:
Just to close the loop on the wcps slope issue... I ran a slope-only workout where for each interval I targeted a pre-selected power and tried to maintain a consistent cadence and speed (as best as I could). Here is the workout file:
```
<workout_file>
  <author/>
  <name/>
  <description/>
  <sportType>bike</sportType>
  <durationType>time</durationType>
  <tags/>
  <workout>
    <SteadyState Duration="120" Slope="0.0"/>
    <SteadyState Duration="120" Slope="5.0"/>
    <SteadyState Duration="120" Slope="10.0"/>
    <SteadyState Duration="120" Slope="0.0"/>
  </workout>
</workout_file>
```
Next I calculated power based on slope and trainer speed for comparison with trainer power. For calculated power, I matched all of the parameters that were used in Flux - weight, crr, air resistance, wind speed, and wheel circumference. There are some mismatches at the interval boundaries due to differential delays and smoothing, but the match was excellent overall, especially after reaching stasis within each interval.

[image]

So I think as far as slope mode goes, it's a rap! Thanks @dvmarinoff.

Also, one very minor thing - in control.js you have units for wheel circumference as "m". Should it be "mm"?

I'm still getting the ble dropouts, and I note that from a comment (//) in wcps.js you have questions about whether at least one of the 1-sec delays are necessary. I'm willing to help with anything more if needed... I continue testing on the dropout issue.

PS: a 2m/sec default wind speed really sucks! :-)

dvmarinoff, 2022-02-06:
Great Job!

Thanks for taking the time, producing and editing the logs!

I am working on fix for the problems you've mentioned. Will push them on Monday together with a first version of auto reconnects. And also a debounce function that hopefully will take care of most `GATT operation already in progress.` errors.

dvmarinoff, 2022-02-06:
About speed, Flux just displays the data that's coming from the device. It is not trying to calculate speed from power.
Wahoo trainers however have this option baked in. It can be selected from the wahoo app. Else most other trainers including my Tacx report flywheel speed which is mostly dependent on cadence and definitely not power. [Here](https://www.youtube.com/watch?v=kkLbcWIh9LY) is GPLama explaining it much better than I can.

The first issue I opened in this repository is about that. Modeling speed for ERG and Slope mode, but most people seem to be uninterested in virtual speed and distance, so I didn't push any effort into this feature.

GCuser99, 2022-02-06:
Right I think I get the concept with regard to erg mode. But in sim mode what happens? Does the cadence data get sent to my trainer, and then based on that, flywheel speed, power, and sim parameters, my trainer calculates speed? Because the speed and power stored in the fit file, along with the sim parameters and implied slopes are consistent with the standard cycling physics model.

I made the comparison above thinking that if Flux was not properly encoding the sim parameters, then there would be a mismatch between my calculated power from fit file speed, and the fit file trainer power. In fact the two were consistent, thus I (hopefully rightly) concluded that Flux is sending the correct sim info to trainer... Is that not correct?

Thanks for your patience...

dvmarinoff, 2022-02-07:
My guess is that the Kickr doesn't deal with cadence at all. They get speed from the power model, which needs just the simulation parameters and measurements from the brake + some calibrated friction. Maybe that's the reason they didn't had cadence.

Your experiment is great on verifying that Flux sends the right parameters, but I don't think it will replicate the result on a Tacx trainer which is kinda perpetually stuck in flatland.

GCuser99, 2022-02-07:
Thanks.

BTW, did you notice the impact of that 2m/s headwind in my power chart, especially on the flat intervals (compare lowest curve with the actual)? I think that is why I ["perceived" that the speed was low](https://github.com/dvmarinoff/Flux/issues/100#issuecomment-1030671214) for my pedal effort!

Can this issue be closed now?

closed 2022-02-08 by dvmarinoff

dvmarinoff, 2022-02-08:
Just pushed an update 0.1.27 for development version. A lot of changes made so good chance of some errors getting through my testing.

- Default wind speed is set at 0 m/s.
- Weight widget is hooked up.
- Weight is being send when establishing connection with a trainer.
- Debounce function with 1000ms will take care of errors resulting from rapid inputs (manual ERG/Slope).

GCuser99, 2022-02-09:
I just completed a short test of Flux dev version using Kickr 2016, Garmin cadence and HRM sensors, and Chrome on Windows 10. Worked with no problems! I'll try a longer "real" workout later...
```
<workout_file>
  <author/>
  <name/>
  <description/>
  <sportType>bike</sportType>
  <durationType>time</durationType>
  <tags/>
  <workout>
    <SteadyState Duration="60" Power="0.75"/>
    <SteadyState Duration="120" Slope="5.0"/>
    <SteadyState Duration="60" Power="0.5"/>
    <SteadyState Duration="120" Slope="3.0"/>
  </workout>
</workout_file>
```
```
db.js:241 start db
views.js:12 start views.
connectionManager.js:9 start connection manager
index.js:22 start app.
idb.js:27 :idb :open :db 'store' :store-name 'session' ...
web-serial.js:214 :serial :ant-not-found
getKnownAnt @ web-serial.js:214
idb.js:93 :idb :getAll :store 'session' :success
models.js:462 :idb :restore 'session' :length 1
idb.js:93 :idb :clear :store 'session' :success
service.js:78 :rx :characteristics  Object
web-ble.js:163 Notifications started on cyclingPowerMeasurement: 00002a63-0000-1000-8000-00805f9b34fb.
web-ble.js:163 Notifications started on wahooTrainer: a026e005-0a7d-4ab3-97fa-f1500f9feb8b.
control.js:475 1644433516109 :rx :wcps :response 1,32,2
control.js:245 1644433517060 :tx :wcps :circumference 21050
control.js:475 1644433517249 :rx :wcps :response 1,72,1,0,58,82
control.js:356 1644433518259 :tx :wcps :sim 72, 0.004, 0.51
control.js:475 1644433518927 :rx :wcps :response 1,67,1,0
control.js:304 1644433519521 :tx :wcps :windSpeed 32768
control.js:475 1644433519650 :rx :wcps :response 1,71,1,0,0,128
common.js:261 rateAdjuster :on pm :tsAvgDiff 1087.15 :result 2
common.js:83 maxRateCount: 2
common.js:83 maxRateCount: 2
service.js:78 :rx :characteristics  Object
common.js:83 maxRateCount: 3
web-ble.js:163 Notifications started on speedCadenceMeasurement: 00002a5b-0000-1000-8000-00805f9b34fb.
service.js:78 :rx :characteristics  Object
web-ble.js:163 Notifications started on heartRateMeasurement: 00002a37-0000-1000-8000-00805f9b34fb.
db.js:81 Object
db.js:122 :zwo set target cadence 0
control.js:22 1644433550159 :tx :wcps :power 150
control.js:475 1644433551030 :rx :wcps :response 1,66,1,0,150,0
common.js:261 rateAdjuster :on cscs :tsAvgDiff 812 :result 3
common.js:83 maxRateCount: 3
common.js:83 maxRateCount: 3
watch.js:277 backing up of 60 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:356 1644433611172 :tx :wcps :sim 72, 0.004, 0.51
control.js:475 1644433612229 :rx :wcps :response 1,67,1,0
control.js:194 1644433613065 :tx :wcps :slope 5
control.js:475 1644433613308 :rx :wcps :response 1,70,1,0,102,134
watch.js:277 backing up of 120 records ...
idb.js:93 :idb :put :store 'session' :success
watch.js:277 backing up of 180 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:22 1644433731178 :tx :wcps :power 100
control.js:475 1644433731391 :rx :wcps :response 1,66,1,0,100,0
watch.js:277 backing up of 240 records ...
db.js:122 :zwo set target cadence 0
idb.js:93 :idb :put :store 'session' :success
control.js:356 1644433791171 :tx :wcps :sim 72, 0.004, 0.51
control.js:475 1644433791872 :rx :wcps :response 1,67,1,0
control.js:194 1644433792881 :tx :wcps :slope 3
control.js:475 1644433794031 :rx :wcps :response 1,70,1,0,215,131
watch.js:277 backing up of 300 records ...
idb.js:93 :idb :put :store 'session' :success
watch.js:277 backing up of 360 records ...
watch.js:53 Workout done!
watch.js:38 Workout done!
idb.js:93 :idb :put :store 'session' :success
db.js:122 :zwo set target cadence 0
watch.js:277 backing up of 364 records ...
idb.js:93 :idb :put :store 'session' :success
control.js:22 1644433922453 :tx :wcps :power 150
control.js:475 1644433923271 :rx :wcps :response 1,66,1,0,150,0
```

dvmarinoff, 2022-02-09:
Thanks! Would have been Impossible without your help! Let me know later how the long session went.