dvmarinoff/Auuki #109: Timers way off

ph4r, 2022-03-19:
[FluxDiag.zip](https://github.com/dvmarinoff/Flux/files/8309645/FluxDiag.zip)
I started this 90 minute workout from vo2fast just as I started the Tour of Watopia event in Zwift. I kept Zwift in the foreground for a long time, I was tunning 0.1.29 Flux in Chrome 99 on Windows 10 using BLE Connections. The first interval ended promptly at 5 minutes as it should, but the 20 minutes interval that followed lasted nearly 40 minutes, and the interval and elapsed timer in Flux were very wrong. The 60 minute interval that followed lasted about 62 minutes which isn't great, but I could deal with it for this low-intensity workout.  I loaded the workout, set my ftp and weight and pressed the play with a cirlce around it button and didn't press anything else in the interface until stop at the end. Here is the same workout as recorded by Zwift [https://www.strava.com/activities/6849027852](url).

dvmarinoff, 2022-03-19:
Hi and thanks for trying out the app.

In general the browser tab wont’t be able to sit in the background for a long time, maybe 5-10 min depending on the OS. At some point it will get suspended and the whole app along with the timer will freeze at this point in time. There might be a workaround by using the esoteric WebWorkers, but they tend to be limited and complicated. Will do some research and will give it a try.

I typically use Flux on my Android phone, and only cycle to the Zwift/RGT Companion App for a couple of minutes if I really need them (they don’t do much that the keyboard shortcuts can’t). Another option is to run Flux and Zwift in a split screen, which works for me on Mac and Android, but I don’t know what’s the state of this on Win10.

dvmarinoff, 2022-03-20:
@ph4r 

Did some changes, by moving the Timer interval to a WebWorker, and my initial testing looks hopeful. The app was able to stay in background with RGT running in the foreground. How long is yet to be determined, but on MacOS the simulator script did a whole 1+ hour workout without issues and I had time to do only a small 25 min session today. Will need more time to confirm it works on Win10 and Android, I just had to ride some long sessions on each platform no other way around it.

Keeping the foreground app in windowed mode, works a bit better than having it in fullscreen mode, in which case some minor things like the painting of the live power graph are glitching.

You can try the latest work with the [developer version](https://flux-devel.vercel.app/).

ph4r, 2022-03-21:
Awesome, thanks for the update!

dvmarinoff, 2022-03-25:
Thanks for the feedback!

I am trying to unpack this, please correct me if I got it wrong.
- you enter new FTP,
- start a workout,
- enter new FTP, at interval 5
- resume workout, but its in interval 7

I know I am a bit behind on writing the manuals and How-To's so lets start with all the buttons:

___1_____2____3_____4____5_____6
[image]

1. the play with a circle starts the currently selected workout, and also the clock and recording if it ain't running already, if the User wants to just start a workout pressing this is enough,
2. the play button is used to start/resume the clock and recording, shortcut: 'SPACEBAR' key
3. pause button, shortcut: 'SPACEBAR' key
4. lap / skip button, this will create a new Lap in free ride, or skip to next interval in workout, after the workout is over the app switches automatically to free ride, it can also be used before the workout is started, shortcut 'L' key
5. stop, to see the stop button the User needs to first use the pause button,
6. download workout, will be available after the recording is stopped

Other shortcuts are: 'e' for ERG mode, 'r', for Resistance mode, 's' for Slope/SIM mode, 'UP'/'Down' arrow keys for adjusting the current active mode target. 

I am a bit confused by your description, the app doesn't have auto-pause functionality, and I couldn't recreate the play with a circle button not starting the workout or clock issue. FTP change is also dynamic you can change the value anytime, and it is not suppose to skip an interval, but it also want update the current target, only the remaining ones. 

There are multiple in progress features on the development version the new big one is the Course simulation. This one got multiple updates for almost every day of the current week, and it's possible that something there is causing an issue, but my first attempts couldn't recreate the described problems, will keep trying, but my best current explanation is that there was some mix up with the short keys or the meaning of the buttons.

As for workout of the day sync, this feature would require at least a server and maybe user accounts for some platforms, which means more complexity to manage including legal one (like GDPR) plus some costs for server time and databases. Until I figure out how to monetize this project this feature won't be available.

Again, thanks for the feedback it is very valuable!

ph4r, 2022-03-25:
Your description identified another issue, at the end of my workout it
didn't go to free-ride, it stayed at the same ERG Power and the Interval
Time started counting up instead of counting down, and the last interval
was still highlighted in the workout view at the bottom.
It is entirely possible that the "didn't start workout" was triggered by
your use of the spacebar; however, Zwift App was in focus when I pressed
the spacebar to launch a power-up and the timers were all at 0 when I
flipped back to the webpage.
When I say auto-pause, I stopped pedaling on my Tacx Neo 2T (in case that
matters) connected via BLE and when I switched from Zwift (Full-Screen via
Borderless Gaming, Windowed App Setting, this works best when I stream my
races and I like to set and forget) to the browser the Play button was
available when I started spinning my feet again against low resistance
(like sim 0%) for the straight chain line gear that I was in. So, while
still watching the browser I pressed play.

No worries on the workout import, I'm just looking for opportunities to get
my VO2Fast workouts loaded into your site or Zwift easier than the many
clicks I perform now.

I have many other features requests for you, but I think course simulation
(I assume to output faux-GPS data so that those who use this app only have
a better experience on Strava et al) is likely a good high priority item,
it was huge for when I ride GTA V Bike Mod.  Relatively simple things like
having an indicator for where in the interval you are graphical, to
complicated things like making a compact view mode so that I can use TR-Z
<https://zwifthacks.com/tr-z-move/> and put the minimum data as a lower bar
on my screen as others do with TrainerRoad. And items in between, like
making the mouseover tooltips be replaced with a dialog with text big
enough I can read (I like to look at the Power Targets for upcoming
intervals) but I can't read the small text you have now and remember what
workouts I've loaded into the local cache between sessions. During Base
weeks, I do the same 3 workouts repeatedly for example.

No matter what, thanks for making, publishing, and sharing this great start
of a solution.
Wesley Stocker

ph4r, 2022-03-27:
I did my 30 minute day-after-race recovery ride with the dev build today,
and the start and interval changing all worked as should; however, the end
is still not quite right. When the workout ended, the graphics stayed on
the last interval but the interval timer counted up, and ERG stayed
engaged, I tried shifting gears drastically to verify I hadn't just gotten
into the rythm.  I pressed Pause and the timer stopped counting but ERG
stayed engaged, I pressed Stop and the timers reset to zero but ERG stayed
engaged.  All this time I have continued pedaling, and a little while after
pressing stop I pressed download and then BAM! ERG Mode disengaged and my
legs went flying due to being in a tiny gear ratio and power dropped as
expected.  Just in case it has crossed your mind, I pair into Zwift as ANT+
and always unpair Controllable there when using another app for structure.

[image: 2022-03-27 (2).png][image: 2022-03-27 (1).png][image:
2022-03-27.png]
Thanks for the good work,
Wesley Stocker

dvmarinoff, 2022-03-28:
> Your description identified another issue, at the end of my workout it
> didn't go to free-ride, it stayed at the same ERG Power and the Interval
> Time started counting up instead of counting down,

Yes, that is exactly how it should work, the app will stay in the last active mode at the last known target, but since the structured intervals are over, the clock will count up to indicate open ended laps. Stop will reset the interval counter, mode and target to the beggining of the workout.

- interval time graphical indicator,

Not too hard to make, but in my opinion it only duplicates information already available and creates clutter. If more people request it I will try to come up with something, but only for large enough screens.

- compact view mode,

In November last year I refactored almost all of the styling and views to bring better responsiveness and cover device screens from tiny phones to 4K monitors. One of the nice to have features was Dock mode (very short + very wide). The problem here is that MacOS and Windows for some reason restrict the minimum height of the browser window to about 300px. To be viable it needs to fall down to like 140-160px. So this remains an option only for linux (Ubuntu) where the User can resize the browser window to whatever size. There are 2 possible workarounds for this, but both have some drawbacks:
1. Creating installable app with something like Electron,
2. Creating installable PWA by adding a service worker,

Since this will also be a pretty solid solution for the ‘use app in background’ issue, I’ll put some time in to investigate, and create a demo. Here is how the current design looks in the device screen simulator (1600x150px):
[image]

- Better display Interval information with mouseover tooltips,
I’ve been wanting to solve this one for a long time, but I am seriously stuck on the design of the whole feature. I’ve tried many approaches and I am not happy with any of them, the current one with the tooltips is just the one I dislike the least. Since this feature is still in design phase I didn’t put any effort to make it responsive to changing screen sizes. Put I agree that it’s annoying will give it another try soon.

ph4r, 2022-03-28:
Thanks for all this back-and-forth.

Please consider a way to jump out of ERG at the end of the workout (I guess
I'm just too used to Zwift or my Wahoo Elemnt Bolt which drop to Sim 0 at
the workout end) somewhere in the backlog (pressing stop even if not an
option or changed default behavior).

I like your Dock Mode, I'll play around with if I can get  Proof-of-Concept
working for this in Chromium on Windows.  I have a few simple ideas (set
width percent on workout frame), and will also see if I can just put your
page inside a fixed-size frame.  I see that when I go short and wide now,
it is responsive but I don't get all the information (and the workout is
even cut-off), but when I make it tall and skinny I get all the
information. There is a trigger at width 1024 where the center stops
vertical scrolling but then drops the other elements.  This looks like an
opportunity in your grid area definitions as changing grid-area,
grid-column-start from 2 to 1 displays almost as your Ubuntu screenshot for
me [except the graph is overlapping the connections (-1) object, and the
targets are horizontal instead of vertical].
[image: image.png]

[image: image.png]
[image: image.png]

ph4r, 2022-03-29:
No edits to your CSS... I made a bookmark as:
javascript:(function(){window.open('https://flux-devel.vercel.app/',
'','width=1920,height=150');})();

Windows 10, Chrome 100

I'll try this setup tomorrow for actual use, today I clicked between the
sessions.

[image: 2022-03-29.png]

dvmarinoff, 2022-03-30:
Nice one, haven't thought about that!

Here is how it looks on MacOS (27'' monitor) with RGT:
[image]

BTW I can't see the attached images on github or in gmail, they appear as only as text: `[image: image.png]`.

dvmarinoff, 2022-03-30:
Added some changes to the developer version:

- responsive tooltips, font size is a little bigger and scales with the base font size of the workout list section for each media query 
- tooltips now show up also for the list of workouts and courses,
- workout/course graphs are now fully calculated with javascript,
- workouts/course graph will respond to window resize event, which itself is debounced on 300ms,
- resolved some style clashes between Dock mode and Landscape phone media queries
- correct setting of initial altitude on course change,

Next I am going to put some time into refining Dock mode, if you have ideas and suggestions now is the time to share.

ph4r, 2022-03-30:
Great, I'll check those out with my next workout!  I guess I should look at
your fit export Course soon then as well. I think the tooltips are still
about half the height they should be, maybe match the size for the timers
font.  You also have the problem I've seen on other websites, on big
intervals the tooltip is cut-off since you anchor it to the top end of the
interval, consider to anchor vertically to the center of the available
height or top-height drawing.

I'd like more screen real estate, so I would love it if you removed the Lap
Power, and Slope.  Slope should be in the Target box, perhaps use a suffix
to make the state clear 'W' '%R' '%S'? Equally, apply the extra space to
the Current Power Graph and the Workout Display.
The connection dots are perhaps a little confusing since the top dot only
does controllable and when only that is paired the other data perhaps
doesn't populate?  To date, I go into settings and pair all 4 lines each
time (it frequently says -Paired after the device, but if I don't go into
the menu it won't start the communication now).
I almost missed the 'Are You Sure' Screen at Workout stop today since it
appeared in the selected theme centered in the available frame, consider
using the alternate theme for this to make it stand out, I thought the
download button wasn't implemented at first until I finally noticed the
pop-up.
Navigating the available Workouts (or as I do each time, scroll to the
bottom and load the custom workout prescribed and downloaded from
vo2fast.com, this could later be a cloud icon if needed ) was very hard to
see, consider splitting the data into columns, preview on one side, and
names / category / length on the other instead of displaying the preview
under the workout when selected.  However, I would also request that you
consider re-using the action bar, instead of having start workout and start
recording that you have Default Workouts (maybe a dumbell icon), Custom
Workouts (upload symbol), and ride report (bar graph icon).

[image: image.png]
[image: image.png]

dvmarinoff, 2022-04-05:
just opened a specific [issue](https://github.com/dvmarinoff/Flux/issues/112) for improving Dock mode, will close this one and move the discussion there

closed 2022-04-05 by dvmarinoff

ph4r, 2022-10-11:
Erg control stayed in tact or is that handled 100% in the trainer itself?

ph4r, 2022-10-11:
[image: 2022-03-30 (3).png]
Today's workout looked great - glad it didn't lose timing on that 60 min
long interval!

dvmarinoff, 2022-10-11:
Hey, @ph4r

glad it worked, did you had the app in the background for the whole duration?

BTW I can't see the attached image can you edit the comment with the Markdown syntax: `[file_name](url)`