dvmarinoff/Auuki #112: improve Dock mode

dvmarinoff, 2022-04-05:


dvmarinoff, 2022-04-06:
@ph4r, just letting you know I added a dock mode button in the latest developer version. It's at the settings page in place of the theme button (I stopped supporting multiple themes long time ago anyway). It will open a new browser window in dock mode, and will place it at the top. Couldn't place it at the bottom, because the bluetooth search popup is getting cut off, and I don't have much control over it. Here is how it looks on Mac:
[image]

I did some experiments with installable PWA, but the thing still refuses to resize bellow 300px height, both with js code and manual. The minimum that I got is triggering only Landscape mode for phones:
 
[image]

ph4r, 2022-04-06:
> Reordered some of the widgets and changes the background-color. I am
> thinking of moving all the buttons from the fixed right section to the
> bottom row, but can't make my mind what should go where.
>
> [image: Screenshot 2022-04-06 at 17 00 46]
> <https://user-images.githubusercontent.com/7523832/162001156-c197aa1f-c785-4899-a01a-e0d34f447fc2.png>

Ok, I'll take a look. I have two more scheduled [tempo] workouts this week
before 2 races this weekend!  I also noticed on my last workout that it
happily switched me to Slope 0 at the end, the only minor concern is that
the last interval of the workout remained highlighted, otherwise this was
great.

dvmarinoff, 2022-04-06:
No worries, I haven't done much yet, just keeping you in the loop.