dvmarinoff/Auuki #113: fix course graph in dock mode

dvmarinoff, 2022-04-07:
In dock mode choosing a course results in a home page graph thats way too long and off screen.

TClin76, 2022-04-08:
[image]
It works well for selecting a course in dock mode on my computer  😄

dvmarinoff, 2022-04-11:
Yes, thanks it's fixed now. Although the strange gap in the bottom row (right corner) in your screenshot is not supposed to be there. I can't replicate this, can you share on what OS, browser and screen size was this taken?

TClin76, 2022-04-12:
I ed it in my office, and here I am using Win11 with a screen resolution of 1920x1080.
Browser is Chrome.

TClin76, 2022-04-12:
Are you talking about the gap as red circle?
[image]
That is because I drag the window to make it larger and then drag it back to normal size. 
After doing that, it comes out the gap.

closed 2022-05-06 by dvmarinoff