dvmarinoff/Auuki #114: Trimmed Interval Power

ph4r, 2022-04-08:
I did my Openers Warm-up today which has 3 short spin-ups which were intended to go to 1010W (as displayed in the Intervals Chart), but Flux only set the target to 800W? My FTP is 287W and my max is ~1450W. I ride a Neo2T (using BLE) which has a 2200W Max.  800W doesn't seem to be calculated from any of these values, so I assume it is a hard-coded limit presently, please greatly increase this limit or make it configurable.

[image]
[FluxDiag2.zip](https://github.com/dvmarinoff/Flux/files/8451275/FluxDiag2.zip)

PS - I missed the 2nd Spin-Up because I was hunting around in the UI and my training plan trying to confirm / figure out why it wasn't as expected. See ZWO and Activity FIT from this session in attached ZIP.

dvmarinoff, 2022-04-08:
TLDR: I extended the default fallback to be 4000W as specified in ANT+ FE-C.

The bluetooth FTMS protocol defines a Supported Power Range characteristic,
which gives min, max and increment values.

On the other hand ANT+ FE-C requires only that the power target be in the range 0-4000W in resolution of 0.25W.

The Tacx NEO uses a custom protocol FE-C over BLE which connects over BLE, but speak ANT+ FE-C.

Thus it falls back to the 800W default max value.

But the Tacx NEO uses some unique technology and is maybe the only trainer capable of actually maintaining higher power values in ERG mode.

Most trainers say they do up to 800W, but in my experience they start to fail above 400W some even only above 300W.

The User would have to switch to what is basically a sprinting gear to get more out of them in ERG mode,
and at that point for most trainers accuracy goes out the window. And having to change gears kinda defeats the purpose of ERG mode.

Moreover for short intervals ERG often takes more time than the whole interval to just stabilise resistance.
Specifically to tackle those issues I added slope mode intervals.
Most trainers will react much quicker to simple slope change than to a new ERG mode power target.

For me setting slope to about 8-10% gives enough resistance to get a sprint effort in small ring/middle of cassette gear.

Anyway ...

Thanks for catching those corner cases!

ph4r, 2022-04-08:
Yeah, the no flywheel is very cool but on rapid Power Changes, it still
takes planned effort to get ahead of the change still (I start my spin-up
2-3 seconds before the interval start).

I'd try your slope idea, maybe you could add an Openers Workout that
implements it, or I look forward to a future where Flux is integrated with
VO2Fast or another no/low-cost training plan generator and there could be a
tailored interface.

dvmarinoff, 2022-04-11:
Attaching here edited version of the opener workout where I adding 14% slope targets for the 3 sprint intervals.

[30min-Opener-slope-mode.zwo.zip](https://github.com/dvmarinoff/Flux/files/8462675/30min-Opener-slope-mode.zwo.zip)

Basically this:
```
<SteadyState Duration="6" Power="3.52" />
```

Becomes this:
```
<SteadyState Duration="6" Power="3.52" Slope="14" />
```

Since Slope targets are a non-standart feature of .zwo, no existing editor supports it. Some time ago I asked a couple of the existing open-source ones, but got no reply. I am thinking of adding a simple workout editor in the 'excel' style of TrainerDay. Might give it a go this week. Closing this issue.

closed 2022-04-11 by dvmarinoff

ph4r, 2022-04-22:
I finally got around to using this today, I can see that it ramped in power, but for these short intervals still didn't get high enough.  However, what I would like to report is what seems to be a strange side-effect of flipping out of Slope Mode.  It appears that your speed calculation does not revert to being at 0% as when I left the spin-ups my speed was like 4-5kph instead of 25kph+
[workout-22-04-2022-at-06-38h.zip](https://github.com/dvmarinoff/Flux/files/8540273/workout-22-04-2022-at-06-38h.zip)

dvmarinoff, 2022-04-26:
Thanks for the Report!

I'll see if there is convenient way to auto-correct this without braking ERG/Slope for courses.