dvmarinoff/Auuki #115: add Workout Editor

dvmarinoff, 2022-04-15:
Workout editor:

- [ ] User can create a workout, executable by Flux
- [ ] User can save the workout
- [ ] User can export the workout in .zwo format
- [ ] User can export the workout in .fit format
- [ ] User can use the custom slope attributes to set slope targets

- [ ] User can add a new row by pressing the add (+) button
- [ ] User can remove the last row by pressing the (-) button
- [ ] User can save the workout from the save button
- [ ] User can export the workout from the export/download button
- [ ] User can add a row to a selection group (multiple rows) by using the radio buttons
- [ ] User can duplicate complex intervals by selecting a group of rows and pressing the add button
- [ ] User can remove a group of rows by selecting them and pressing the remove button

[image]

cbourgeois, 2022-04-26:
Nice initiative, I played a bit with it, good job.

Overall I think that the best UI for workout editors out there is the one provided by intervals.icu  :
https://forum.intervals.icu/t/workout-builder/1163

Just typing :

> 6m 50%
> 10m 40-50%

is much faster than fiddling with each value in a table like the one above.
Displaying the graph is also a nice UI feedback, which allows in a glance to ensure the sanity of the values.

I would recommend that design.

At the moment, the biggest limitation with workouts in your otherwise great app is the fact that files workouts are not saved across sessions, and it's not very friendly to have to reupload them everytime.

TClin76, 2022-04-27:
Does Slope have higher priority than ERG If target power and target slope presented in the same interval?

dvmarinoff, 2022-04-27:
@TClin76 Yes, slope has higher priority. To disable it for the interval leave it blank or dash, but 0 would set the trainer to slope mode with 0 grade.

dvmarinoff, 2022-04-27:
@cbourgeois

The text declaration syntax is an interesting option, thanks for sharing! Do you know what is the general feedback on this approach from intervals.icu?

> workouts are not saved across sessions
The app is fully static, which leaves the option to choose between the browsers' IndexedDB and LocalStorage. I've been thinking on the problem, but I can't make my mind on how to approach it. IndexedDB is frankly giving me nightmares, and I don't know if I can trust LocalStorage for storing User data of that size. I am thinking also of adding some sort of bulk import/export solution that will allow for storing the workouts on the local user device and transferring them to another device.

dvmarinoff, 2022-04-28:
Created a poll discussion about the interval definition. You can vote [here](https://github.com/dvmarinoff/Flux/discussions/118)

cbourgeois, 2022-04-28:
> @cbourgeois
> 
> The text declaration syntax is an interesting option, thanks for sharing! Do you know what is the general feedback on this approach from intervals.icu?
> 

Hey @dvmarinoff , the feedback around this looks positive, from what I can gather reading the thread I linked above.
I also personally found it very handy, quick and powerful, this is of course biased by the fact that this I am in the software industry. 

> The app is fully static, which leaves the option to choose between the browsers' IndexedDB and LocalStorage. I've been thinking on the problem, but I can't make my mind on how to approach it. [...]

I understand, this is indeed not simple, perhaps it would be nice to find a way to link to a bank of workouts hosted somewhere else that users could simply load into the app before a workout ?