dvmarinoff/Auuki #119: Syntax of zwo format

TClin76, 2022-05-05:
[image]
Hi @dvmarinoff 
I found that there are some zwo files downloaded from Whatsonzwift use PowerHigh and PowerLow within SteadyState Duration command.
FLUX does not recognize this kind of syntax and only accept PowerHigh and PowerLow within Warmup/Cooldown Duration.

dvmarinoff, 2022-05-05:
Having Low and High attributes on a SteadyState doesn't make any sense to me, but indeed it is working in this unexpected way:
[image]

dvmarinoff, 2022-05-05:
Just pushed a fix for this case to the development version.

My guess is that this situation arises from visual editors transforming SteadyState intervals to Ramps and back.
So if PowerLow and PowerHigh are present on a SteadyState the App will take the higher value from the 2 and use it as Power.

In this example workout all SteadyState will result in intervals of 60s at 0.70% of FTP:
```
    <workout>
        <SteadyState Duration="60" Power="0.70"/>
        <SteadyState Duration="60" PowerLow="0.70" PowerHigh="0.70" />
        <SteadyState Duration="60" PowerLow="0.40" PowerHigh="0.70" />
        <SteadyState Duration="60" PowerLow="0.70" PowerHigh="0.40" />
        <SteadyState Duration="60" Power="0.70" PowerLow="0.70" PowerHigh="0.70" />
    </workout>
```

[image]

dvmarinoff, 2022-05-05:
@TClin76 Thanks for reporting the issue!

TClin76, 2022-05-05:
> Just pushed a fix for this case to the development version.
> 
> My guess is that this situation arises from visual editors transforming SteadyState intervals to Ramps and back. So if PowerLow and PowerHigh are present on a SteadyState the App will take the higher value from the 2 and use it as Power.
> 
> In this example workout all SteadyState will result in intervals of 60s at 0.70% of FTP:
> 
> ```
>     <workout>
>         <SteadyState Duration="60" Power="0.70"/>
>         <SteadyState Duration="60" PowerLow="0.70" PowerHigh="0.70" />
>         <SteadyState Duration="60" PowerLow="0.40" PowerHigh="0.70" />
>         <SteadyState Duration="60" PowerLow="0.70" PowerHigh="0.40" />
>         <SteadyState Duration="60" Power="0.70" PowerLow="0.70" PowerHigh="0.70" />
>     </workout>
> ```
> 
> [image]
> 
> Here are the test cases:
> 
> https://github.com/dvmarinoff/Flux/blob/85612444508da832ab021e76b465c8551ce1305c/test/workouts/zwo.test.js#L549

I modify the zwo file and let powerHigh and powerLow have different value, and then test it in Zwift.
The result will be the average of two values.

dvmarinoff, 2022-05-05:
True that, just checked too it in Zwift! Will update the code.

[image]

[image]

dvmarinoff, 2022-05-05:
Updated the code to take the avg value if powerLow and powerHigh differ:

[image]

dvmarinoff, 2022-05-06:
Merged into master, closing this one.

closed 2022-05-06 by dvmarinoff