dvmarinoff/Auuki #122: Customisation of the interface

dvmarinoff, 2022-06-06:
> Would love to see some customisation of the interface, such as ability to remove fields, increase/decrease field sizes. Like you can do with a bike computer.

@Zhwifty, Thanks for the review and the name suggestion. I allowed my self to move your question to a separate issue here.

Customisable UI can be very complex to develop and maintain and right now I have all my time occupied with developing the backend API and the iOS app.

Which fields would you like to add/remove? I may start thinking on how to design this feature, but I don’t know if I’ll have the time to develop it soon.

Zhwifty, 2022-06-06:
After looking again at the data fields, there probably isn't any unnecessary information.

But if I could assign more space to the graph and less to the text, as well as ability to reduce font size of the text boxes it would transform this app into something beyond anything else available.

I see great potential in this app to be used in landscape mode, as a high level cycling training system, which could cross both indoor and outdoor realms.

The ability to see the intervals in graph form, akin to the way the we see in zwift/trainerday/trainerroad etc is very motivating. 

I think what I am trying to say is that the app has great potential to be used in landscape mode, to display the essential training metrics which when paired with a large graphical display of the training session would set it apart from anything else I have come across as either an app or a dedicated cycle computer.  

My suggestion is perhaps a bit off piste and probably would require ability to use ANT+ for those that like to record speed, distance and routes with a bike computer. 

Anyway, thanks so much for your work so far.  I have no knowledge of coding programming so I apologies if what I suggest is technically not feasible/too complex.

dvmarinoff, 2022-06-07:
@Zhwifty made some changes to the developer version [https://flux-devel.vercel.app/](https://flux-devel.vercel.app/) of the app. It has 4 rows and all of the fields. We can keep iterating on it if you have any suggestions or comments:

[image]