dvmarinoff/Auuki #126: Elevation Gain in Workout Display?

GCuser99, 2022-08-03:
Hi @dvmarinoff,

I'm still using Flux's distance mode to simulate hill climbing - I love it! I noticed that the only important stat that is missing from the display is elevation gain. I also noticed that the "Target" stat is unused in distance mode, at least the way I'm running it. Wondering if it would make sense to use that space in the display to report elevation gain...?

Thanks!

dvmarinoff, 2022-08-04:
Hi @GCuser99,

Will have to think about it, the difficult part is figuring out where to put it in the current interface so it makes sense.
I’d like to keep the Target field, because some may want to switch between ERG and SIM mode while riding courses.
I’m thinking maybe somewhere inside the elevation graph, or making the Slope field switchable between Elevation gain and Slope.

GCuser99, 2022-08-04:
Oh I forgot about the Slope/Erg mode switch...

I wonder if "Target" can auto-switch to "Elev Gain" when mode is Slope and then back again when mode is Erg? Or is that too messy to implement?

I'm using a fit file workout and really would like to see Slope at all times. If you were going to make something user-switchable I think it would make more sense to make the cumulative stats distance and elev gain switchable. 

Thanks again.

dvmarinoff, 2022-08-10:
@GCuser99 just added current Elevation and Ascent to the top part of the course graph of the developer version.

> If you were going to make something user-switchable I think it would make more sense to make the cumulative stats distance and elev gain switchable.

Agree on that. It's just too many fields and too many use cases. I am thinking of adding ability for the User to configure some of the slots for the fields. But that's a bit complex to implement. It will need a UI for the configuration and then configurable slots, both of those functioning on Mobile and Desktop.

NOTE: the values are in meters, will soon add Imperial conversions.

GCuser99, 2022-08-10:
Cool - thanks for that. If I do multiple laps (consecutive repeats of workout without stopping) will the Ascent value keep accumulating?

Anyway I understand the complexity that comes from accommodating every use case and appreciate you at least considering what makes sense, given your main development objectives. Thanks again!

dvmarinoff, 2022-08-15:
@GCuser99 it should accumulate the ascent over lap. I did 2 loops of the short Borrego Springs Course to check this and it's working.

I am planning for this winter to make the Course graph a 3D thing and play with some very simple generative graphics. I am also thinking the Live Power graph is either going away or shrinking a lot in size. Maybe the Ascent and Current Elevation fields will have more space in the new design.

TClin76, 2022-08-16:
> @GCuser99 it should accumulate the ascent over lap. I did 2 loops of the short Borrego Springs Course to check this and it's working.
>
> I am planning for this winter to make the Course graph a 3D [...]

An idea, maybe put the Heart rate graph together with power graph? Something like what Zwift does.