dvmarinoff/Auuki #139: Feature request

ahaige, 2022-12-24:
Great project, thank's a lot for your work!
It would be helpful to show the content of textevent tags from zwo files. Additionally target values for cadence would be nice.

dvmarinoff, 2022-12-29:
Hey, thanks!

Will put text events in the backlog, but not sure when I will have time to implement those.

About cadence targets I [looked up](https://github.com/dvmarinoff/Flux/pull/74/commits/a9a225780c4947fe1f83dd481e414c7a4e90312c) and it seems last year I added limited support only for strict cadence targets on SteadyState and IntervalsT tags, but cadence ranges are not supported, mostly because it's a bit harder to find space on mobile devices in which to put all the info and this will require a lot more effort and planning on how to design another custom UI widget without creating needless clutter.

ahaige, 2022-12-30:
Strict cadence targets are fine for me. I didn't check that its already implemented and visible in the UI widget, if set.

closed 2023-11-20 by dvmarinoff

karlacio, 2024-04-12:
First of all thanks for this wonderful app!

Sorry if I'm missing something.

I was looking if support for textevent messages was already done and I found a reference in this closed request.

Using the app I see no textevent. Are they already implemented?

Thanks.

dvmarinoff, 2024-04-12:
Hi @karlacio 

Textevent messages are not implemented. I am not a big fan of those, so I am not sure if I want to add them.

karlacio, 2024-04-12:
Thanks for your reply, I understand.

I'm new to workout training.
I found sometime useful Textevent messages for:
- cadence (if not in zwo file)
- position (standing, seated)

So I was asking just for that.

If one day you'll implement them plese add the option to have a sound alert when a new message appear.

Thanks again,
Carlo.