dvmarinoff/Auuki #140: ANT+ HR Connection Status + Incorrect Workout Distance

shimeike, 2023-01-11:
First of all, thanks so much for this awesome project. Having recently started using a Kickr smart trainer, I was quite surprised to find the very limited number of free options to automatically control the trainer to do custom structured workouts.

I have noticed the following 2 minor issues: (Using Google Chrome on Linux with Wahoo Kickr controlled by Flux over BLE and ANT+ HR monitor connected via Garmin Ant+ USB dongle.)

1. The "HRM" connection indicator on the workout screen remains Red, even though HR is being received and displayed. I would expect it to be Green, as it is if I connect to my Garmin watch broadcasting HR over BLE.

2. I have the Kickr set to "Erg Mode Speed Simulation" as I'm using erg mode in Flux. I'm also recording (not controlling) the workout on my Garmin Edge head unit. The speed readings between Flux and the Garmin are not identical, but close. However, the distance displayed in Flux is approximately double that on the Garmin, which seems to be an error. 

Neither are major topics - the power control/display and HR/cadence display are the most important and they are correct.

dvmarinoff, 2023-01-13:
Thanks for the feedback! Those are realy valuable. 

The ANT+ functonality is experimental and still in early development. Having ANT+ in a browser is pretty hard to achieve and frankly I've been stuck on an issue developing this feature. If you know anyone who understands USB and web and is willing to help it would be great. 

About the distance thing I'll do a test over the weekend and investgate. Flux has an option in settings where you can switch between speed reported from the trainer and speed estimated from the power. Double overestimation although sounds like maybe an issue with units convertion (meters per second to km/h or miles and back). Will look into it.

shimeike, 2023-01-13:
Thanks for the info!

The ANT+ HR monitor connection status is green on the "Settings" screen - just not on the "Home" screen. So I don't think this particular issue relates to the lower level stuff. From my experience, it works great!

Regarding speed - I did another workout yesterday and did not have the "double distance" issue. (Note, the issue was not that speed was double, only distance - which is really strange. I have a fit file from Flux showing a distance of approx. 60km for workout time of under 90min with average speed of approx 25km/h.) Yesterday, I only had the slight difference in speed (between Garmin/Wahoo and Flux), which would be attributable to different power estimation parameters. Maybe it was just a glitch on that day with my setup. I can send you the fit file if it helps?

I had completely missed the speed/power setting. I had it set to 'power', which I guess means Flux uses its own estimation parameters - hence the slight difference between Wahoo/Garmin. I guess 'speed' is what I want.

Thanks again for the awesome project!

closed 2023-11-20 by dvmarinoff