dvmarinoff/Auuki #142: Training Effect/Load won't be calculated (Garmin)

PaddoSwam, 2023-01-25:
I've just finished a small ride, worked great btw., but the downloaded fit file uploaded into Garmin Connect doesn't seem to calculate a Training Effect nor Load.

I haven't figured out what the issue might be but my first hint was to a vo2max/ftp setting that might be missing.

Will investigate this one myself too because open-source! 💪🤓

dvmarinoff, 2023-01-25:
Thanks!

Will look into it.

dvmarinoff, 2023-01-25:
I don't have it on my Zwift recorded rides either. I am starting to suspect it is a garmin device exclusive feature.

Plus this is what garmin connect says:
[image]

PaddoSwam, 2023-01-25:
My Zwift rides imported to Garmin Connect automagically (not sure if push by Strava/Zwift or pulled from Strava by Garmin)

PaddoSwam, 2023-01-25:
Just rode on MyWhoosh and had to manually upload my fit file and that one did get a load an te calculated, I'll compare the 2 files!

PaddoSwam, 2023-01-25:
There is quite some differences, but I suspect the session data part and/or device part. Will take a closer look tomorrow/friday. 🤓

[Flux_Half_Chili_Pepper.fit.txt](https://github.com/dvmarinoff/Flux/files/10503681/Flux_Half_Chili_Pepper.fit.txt)
[MyWhoosh.fit.txt](https://github.com/dvmarinoff/Flux/files/10503682/MyWhoosh.fit.txt)

PaddoSwam, 2023-01-25:
> I am starting to suspect it is a garmin device exclusive feature.

It is something that a Garmin devices calculated, Garmin Connect doesn't do the calculations, for me it's a Garmin Forerunner 745 and/or Garmin 530 that calculate the efforts.

dvmarinoff, 2023-01-26:
I see, will investigate more closely tomorrow.

PaddoSwam, 2023-01-26:
I think I fixed it!

dvmarinoff, 2023-01-27:
**Ray’s article says**
- Training Load is calculated on a Garmin device, not Garmin Connect,
- but fit files recored with Zwift, TR, WahooX and Tacx since June 2021 will get Training Load updated on the device,
- They do it by syncing with the device and calculating it there and than uploading it back to Garmin Connect,
- Training Effect was not available at the time (June 2021),

**Few things to note from your MyWhoosh .fit file:**
- The manufacturer is listed as Garmin and the product as Garmin Edge 1030,
- Here MyWhoosh should be using their own manufacturer id and product id or set it to 255 meaning ‘development product/id’,
- So the .fit files recorded with MyWhoosh claims that it’s recorded with Garmin Edge 1030, and this way it circumvents the current limitations

[image]

**I modified the .fit file code to match what MyWhoosh is doing and ran some experiments:**
- Changing the manufacturer field to Garmin
- Changing the Product field to Edge 1030, and Fenix 5 (my watch),
- Setting Serial Number to the one in your file, and to the one of my Fenix 5 watch,
- Adding File Creator message with Software Version field,

**Results:**
- When setting it to Edge 1030
- Garmin Connect sees the file as created by Edge 1030,
- After a sync Training load is successfully updated, both on my Watch and the mobile Garmin Connect app,
- Training Effect though is missing, both on the watch and on Garmin Connect,

[image]

- When setting it to Fenix 5,
- Garmin Connect sees the file as created by Fenix 5,
- After a sync Training load is not updated, but only Recovery time,
- Training Effect is also not there

So having Training Load updated everywhere is a partial success for now.

BTW do you own a 1030 navigation? And did MyWhoosh had you enter Garmin device type anywhere in settings?

PaddoSwam, 2023-01-27:
Hey Dimitar, no I own a Edge 530 and Forerunner 745 both capable of calculating the training loads.

The _Edge 1030 Plus_ was virtually added by MyWhoosh, and so I thought of changing the same thing you did and that seemed to work however my quick edit in Flux code failed so I'll retry right now :)

dvmarinoff, 2023-01-27:
Pushed my changes to the [develop branch](https://github.com/dvmarinoff/Flux/tree/develop).
They are also live on the dev deployment: [flux-dev.vercel.app](https://flux-dev.vercel.app/)

PaddoSwam, 2023-01-27:
Awesome the Training Load is calculated properly and so is recovery time, I think I found my new favorite trainer app!

closed 2023-01-27 by PaddoSwam