dvmarinoff/Auuki #143: Sorry for question, how to run program please?

jimx3D, 2023-01-31:
So I am using Giant Combo BKM4AB with Ant+ and Ble, and Chromium browser. I run Windows 10. I have Garmin USB Ant+. I look in src folder, I see index.html but when I running in Wamp just static page no values changing when sensor moving.

When I click 'settings button', in Chromium is error like this:

 "localhost/:1 Uncaught TypeError: Failed to resolve module specifier "@vercel/analytics". Relative references must start with either "/", "./", or "../"."

What is solution please? Thank you.

dvmarinoff, 2023-01-31:
It depends on your goals:

1. You just want to use the app:
- then go to: https://flux-web.vercel.app/

2. Your want to run it locally: 
- then you don't need vercel analytics,
- you should remove it as a dependency from package.json: https://github.com/dvmarinoff/Flux/blob/3f46db0a482817341cc481d5282b554611851845/package.json#L20
- and remove the import statement and the call to `inject()` from the code here: https://github.com/dvmarinoff/Flux/blob/3f46db0a482817341cc481d5282b554611851845/src/index.js#L8
- then run `npm install`
- then `npx parcel index.html` should startup a local server at `localhost:1234/`

jimx3D, 2023-01-31:
Thank you for reply, but https://flux-web.vercel.app/ is not working for me so will leave for now until project more advanced. Thank you for taking time.

closed 2023-01-31 by jimx3D

dvmarinoff, 2023-01-31:
This project is BLE only, ANT+ is just for Heart Rate Monitors and is experimental feature. The web page requires Google Chrome browser on Windows 10, although it will work with any **modern** browser that has support for Web Bluetooth.

jimx3D, 2023-01-31:
ANT+ is nice to have but really looking really for quicker Ble code than https://github.com/amcolash/oled-bike-computer. OLED working very well with Giant Combo, but so slow to send data, one second per update, is too slow. Nevermind, is no problem really, just personal project for make key press in game. I have oled code working, then I add Python websockets to direct cadence data on to arrow key. Is working, but is slow!! I know this also websockets is slow as well so not just Ble code. Really I think is best to make Ble code also Python but too much for me just now!! Thank so much for helping with quick replies ^^