dvmarinoff/Auuki #145: Support o CycleOps  PowerTap G2/G3 over ANT+

torresweb, 2023-06-19:
Hi,
Please, anyone could give any tip how to include support for my old CycleOps G2 powermeter?  Looks that webant+ project is frozen. I am available to give any information, test and even put some lines of code if some help is provide me. 
Thanks.

dvmarinoff, 2023-06-19:
Hi,

Thanks for checking the projects!
I can't find any info on the internet about this power meter, but if it only supports ANT+ than the only option to support it in Flux is by getting the WebANT+ project done. This one has 2 issues.

First last time I checked there is no practical way to use it on Windows. You'd have to change drives manually to get it working on Windows. And second there is a bug with the code that makes the connection work only for a bit than stop than work again and so on. Now think I know how to fix, but I didn't have the time this year to work on it.

If its viable solution for you, you'd need to be on another OS, but Windows. And than we have to check if this powermeter at least connects and works for a bit in the current version of WebANT+. If not than its using some custom protocol or it is encrypted and I can't do much to supported.

torresweb, 2023-06-19:
My hub: https://www.thisisant.com/directory/powertap-pro-hub1/

dvmarinoff, 2023-06-20:
Nice, it seems to support the official ANT+ Power profile. Will see what I can do over this weekend and let you know when it's time for testing with the physical device.

torresweb, 2023-06-21:
May be this code can help us: https://github.com/8beeeaaat/web-ant-plus
I tried the example code and it works fine with my classic PowerMeter, see the log:

BicyclePowerSensorState {DeviceID: 33686, ..., Cadence: 64, …}
AccumulatedPower: 22495
...
ReceivedAt: 1687354611475
...

dvmarinoff, 2023-06-21:
Thanks I'll sure take a look at the code! I am most curious if they have any solution for the driver issue on Windows.

coetzeeconradcadence, 2026-09-15:
Hi — I've been chasing this exact problem this week for a fleet of ~20 studio CycleOps/PowerTap hubs (older split-battery-cap generation, patents US D533,237 / 7,408,447 / 7,585,258, Hub IDs like 24738). Posting what I've confirmed in case it helps, and in case anyone here has gotten further.

**Hardware split observed:** about 4 of ~20 hubs speak the standard ANT+ Bike Power profile (device type 11) directly and work with any ANT+ app. The rest (majority, e.g. ANT device number 65491) don't respond to type 11 at all, but *do* work with a third-party app (Selfloops) on the same tablet.

**What I found decompiling that working app:** it uses a non-standard device type (8) and non-standard channel params (tx type 3, period 10923, RF 53) instead of the official Bike Power profile — but critically, it also applies a small app-layer XOR/shuffle (3-byte key) to the *received payload* after the fact. That's payload descrambling, not a radio-level ANT network key.

**What I confirmed empirically:** using the same device type/tx/period/RF (8/3/10923/53) on the standard PUBLIC ANT+ network, with correctly-implemented claim handling and search timeout (confirmed firing at exactly the configured duration), and the hub confirmed physically spinning the entire search window — zero reception. Not a code bug on my end; a clean, fully-elapsed search that heard nothing.

**My working theory:** these hubs are pre-ANT+-Sport-update PowerTap 2.4-era units that only ever spoke a private Saris network (this matches an old LW Coaching article about a hardware dongle Saris sold ~2009-2010 to add public ANT+ Sport support on top of the private network). The 4 working hubs in my fleet may simply be units that received that physical update at some point; the rest never did.

Has anyone here gotten further on the actual private network key, or does anyone know if that old Saris hardware updater still turns up anywhere? Happy to share more detail on the channel params/decompile if useful to anyone else hitting this.