dvmarinoff/Auuki #165: Power Matching

enzingerm, 2023-11-24:
It would be nice to be able to control the trainer resistance in ERG mode based on the readings of a separate, more accurate power meter.

I am willing to possibly contribute to this feature myself but as of now I don't have an overview of the code structure, just wanted to open this issue for input and further discussion.

enzingerm, 2023-11-28:
Closing this because I did overlook #92 by only searching for "power matching" before creating my issue

closed 2023-11-28 by enzingerm

dvmarinoff, 2023-11-28:
Hey, this feature has been requested before, but it is going to be very time consuming to be implemented. Also just looking at how many configuration options and toggles exists for this functionality in the TrainerRoad app makes me skeptical it's even possible to create a simple feature that just works. If you have the required theoretical understanding and the time to sit down and implement the feature I'll try to integrate it into the project. My one requirement is using no dependencies though.