dvmarinoff/Auuki #166: Feature: intervals icu integration

NickLewanowicz, 2023-12-01:
Would be super slick if the web app integrated with intervals.icu, ideal scope would be: 

1. select a workout to start from your training plan on intervals
2. Automatically set the workout based on what's defined in intervals
3. Automatically upload your ride metrics to intervals afterwards

Even just the first item here would be great, it doesn't need to be fully driven by intervals. For example, if you could have a section of the website that shows your planned rides synced from intervals and allows you to match it with existing Flux workouts that would avoid users needing to switch between the 2 apps when starting a workout.

Lunchtime0614, 2023-12-23:
Yes I agree with this - Intervals provides an open API that allows you to pull in the workout for that day/week...just need to have that loaded to the workouts screen...or load it on demand with an API call.  I've been looking into this but have no idea where at this point flux would/could read the API response of the workout and then load it/have it selectable.

dvmarinoff, 2024-12-21:
I’ve just released a demo version of the Intervals.icu integration that you can test out at `dev.auuki.com`.

The current list of supported feature:
- syncing of planned daily workouts from Intervals.icu to `dev.auuki.com`
- uploading finished workouts directly to Intervals.icu

You’d need to go to the Settings Tab -> Profile and create a temporary account, the domain `dev.auuki.com` is just for testing and feedback I’d delete the database before the feature goes live on `http://auuki.com` (so you can use a fake email, if you'd like). Once you register and login you’d be able to connect Intervals.icu from the Profile page.

Current known issues:
- password reset is not active for this version of the app

If you are up for it, I’d love to hear your thoughts and feedback.

Cheers and Happy Holidays!

dvmarinoff, 2024-12-28:
OK, the integration is live on Auuki side, I'd appreciate some feedback. Thanks everyone!

closed 2024-12-28 by dvmarinoff