dvmarinoff/Auuki #173: Wrong speed

Gregdimakis, 2024-01-03:
Since Christmas, the app has started showing wrong speed when uploaded to Garmin connect. Also it tries to load a map on Garmin. I noticed that prior to this, the activity had shown as connected with edge 1030 plus, but now there is no connection.

dvmarinoff, 2024-01-03:
@Gregdimakis Thanks for the report!
I did a huge update on the .FIT parser. I see that the speed is not getting converted from k/h to m/s:

[image]

Will fix that issue tomorrow morning. Meanwhile sorry for the inconvenience!

dvmarinoff, 2024-01-04:
@Gregdimakis just updated the app with some fixes. Speed should be working fine now. Also fixed a timestamp bug with the Laps.

closed 2024-01-04 by dvmarinoff