dvmarinoff/Auuki #175: Missing first interval on final surge

Gregdimakis, 2024-01-06:
Thank you for the great work done. Amazing job!

I have noticed that when I synchro my activity from Garmin to Final surge it is always missing the first interval (warm up; not a specific duration). The fit file  seems ok on Garmin. Also, the same activity from different source synchronises ok on Final Surge thus I guess it is something on the produced fit file.

All the metrics seem ok. It is only in the graph and the laps.

Thank you in advance

Gregdimakis, 2024-01-06:
Additional screenshots 
[image]
[image]
[image]
[image]

dvmarinoff, 2024-01-06:
@Gregdimakis Thanks for the report I am looking into it, might take a few days.

On another note, I am about to publish a new update of the .FIT file code. It fixes some mild unrelated issues.

dvmarinoff, 2024-01-06:
@Gregdimakis I just did 2 short rides with both the old (v 0.1.41) and the latest (v 0.1.42 )version of Flux .FIT file logic.
Both pass the  initial checks. Laps (Intervals) are recorded correctly and are displayed correctly in Garmin Connect, Strava and Intervals.icu.

### Garmin Connect:
[image]

### Strava:
[image]

### Fit File Viewer:
[image]

I am going to take a look at Final Surge, now, but it has 2 UIs: Beta and Classic. Do you know which one are you using?

[image]

Gregdimakis, 2024-01-06:
Thank you. I am using the beta version