dvmarinoff/Auuki #181: Zwift hub one Click integration

CarlosMenke, 2024-01-23:
Hey, i just got a Zwift hub one with virtual shifting, and i wanted to ask if you could implement virtual shifting of the cycling trainer, because without it is hard to use with your app?

Thanks in advance!

bttger, 2024-04-06:
As far as I know, Zwift has not published any specification about the Zwift click. So we don't know how to read it (yet).

Wahoo [writes](https://support.wahoofitness.com/hc/en-us/articles/16637692311954-Why-choose-CORE-with-Zwift-Cog-vs-cassette) "If you plan to ride in Zwift, but also in other apps like our SYSTM training app or others, the Zwift Cog will work for most applications, but because the Click only functions in Zwift, you won’t be able to shift gears. This is fine for ERG mode workouts, but for other virtual ride environments, fitness tests, or some sprint workouts, shifting is necessary.  In any case, read on for other considerations."

dvmarinoff, 2024-04-06:
Hey, folks, I've been slowly working on this. In another project we have the Zwift Click commands figured out and the traffic decoded. What remains is the control of the trainer. I am sceptical that Zwift/Wahoo would ever release access to the public, so this would need to be solved in a round about way. In this other project we have explored the idea to use ERG mode to Simulate virtual gears. It's working, but it needs more refinement. 

In the future I'll add support here. There is actually some code in the project already. The biggest obstacle is figuring out the function that matches nicely ERG resistance to virtual gears. And the second thing is the encryption. The Zwift Click uses AEC CCM mode, but the browser has support for AES GCM mode. So I'd have to implement a small encryption library too.
It's a lot of work to do all these tasks, so it will take time, but it's definitely coming.

dvmarinoff, 2025-10-15:
@benjmarshall no updates on that front so far. I'll wait until Zwift makes the protocol public (haven't checked in awhile though). Otherwise I like the idea that someone propose to use the Zwift Click/Play as a way to remotely control Auuki, and this is definately coming this indoor season.

effectpears, 2026-10-18:
The indoor season is starting and the `zwift cog` is widely used. Could we revisit this topic? I'm happy to test things or provide data.

dvmarinoff, 2026-10-18:
Very well, will contact you next week because I need someone to test this.
BTW is anyone willing to help for testing the Zwift Frame / Play controls, and the new Stages SB200 bike over the OpenBikeControls protocol (https://github.com/OpenBikeControl/bikecontrol)?

effectpears, 2026-10-18:
I have the `Zwift Cog and Click Upgrade Kit` with the controls (should be v2).

effectpears, 2026-10-20:
@dvmarinoff https://github.com/thomaschampagne/zwiftboard maybe this is also interessting for you / auuki.

It would be a huge benefit if the clicks worked natively with auuki to also support other operating systems (Android).

dvmarinoff, 2026-10-21:
I got distracted with the OpenBikeControl protocol (https://openbikecontrol.org/), and I think I'll do this one first. This will allow me to figure out the redesign of the control section of Auuki.