dvmarinoff/Auuki #182: Local web server

kurtnolf, 2024-02-17:
Great project ! Is there a reason that this SPA cannot run locally ? I installed a local web server and I downloaded the source . The home page loads fine but I cannot go into the workout or other pages. Any specific requirements to be able to run this on a local webserver ?

dvmarinoff, 2024-02-17:
The project should run locally, it has no external dependencies except for the dev tools (parcel), but you can easily circumvent those. Have you tried the following steps?

```
git clone https://github.com/dvmarinoff/Flux.git
cd Flux/
npm install
npm run start
```

In a supported browser open: `localhost:1234/`

kurtnolf, 2024-03-02:
> The project should run locally, it has no external dependencies except for the dev tools (parcel), but you can easily circumvent those. Have you tried the following steps?
>
> ```
> git clone [...]

Thanks for the instructions. It works fine.

closed 2024-03-18 by dvmarinoff