dvmarinoff/Auuki #186: Power not working

Slinky6267, 2024-03-20:
Power doesn't show on flux or record to file after the newest update, before that it was working fine. 

Controllable light is green and my trainer receives power level from a workout file.

I have tested it on desktop and android, neither of them work.

Trainer: Elite Suito

dvmarinoff, 2024-03-20:
@Slinky6267 Thanks for the report!

Any chance you could send me the logs from the developer console?
[https://flux-dev.vercel.app](https://flux-dev.vercel.app) has the  logs enabled.
You can open the developer console in Chrome/Edge on desktop with right-click -> inspect.
Try to connect the trainer, ride for 20-30 seconds, right-click Save As.

Slinky6267, 2024-03-20:
[flux-dev.vercel.app-1710942460342.log](https://github.com/dvmarinoff/Flux/files/14667372/flux-dev.vercel.app-1710942460342.log)

dvmarinoff, 2024-03-20:
Thanks! Will go over it!

dvmarinoff, 2024-03-20:
@Slinky6267 Hey, I just added a fix for this issue to the dev version at: https:flux-dev.vercel.app.

The issue is, that during the recent changes to the Bluetooth code, I have missplaced the getter for Uint24 values.
And Elite trainers send total distance which is formated as Uint24, so the whole parser was breaking because it couldn't find the Uint24 getter. Anyway, I included this bit back and added some tests here:

 https://github.com/dvmarinoff/Flux/blob/b8bcb3fbfa3aeb87506b190ad901dfff6eb25782/test/ble/indoor-bike-data.test.js#L35
 
 This should cover the use case. But can I ask you to test with the real device, the Elite Suito trainer whenever you can?

Slinky6267, 2024-03-20:
After a quick test it seems to be now working. Thanks for the quick fix!

dvmarinoff, 2024-03-20:
Thanks for the help and sorry for the inconvenience!

Closing this now...

closed 2024-03-20 by dvmarinoff