dvmarinoff/Auuki #188: 0.1.48 - Multiple Issues

Garyrk001, 2024-03-21:
Thanks for all the work on this app.  Been using for several months with no issues. Today with v0.1.48, I had multiple issues.  

Samsung Galaxy S9
Andriod v10, Chrome browser
Wahoo KICKR Snap

Everything seems to work ok with Zwift on phone.
Issues:

1. Trainer in ERG mode, going through a workout, control is not working.  At start, I can pedal at any power output without any response from the trainer to match the workout (does not seem to be in ERG mode).  I also get some very quick changes to resistance levels from the trainer not in the workout for a short period - like it jumps to a steep slope.  Further, changing the power via the +/- buttons has no effect on trainer.  Trainer is paired with app and shows green.  And it is showing power output correctly.  Just not in ERG mode correctly.

2. Cadence sensor not working.  I am able to pair a generic BLE cadence sensor to the app, it shows green, but the cadence is never registered.

Are old versions available?  I can't use the app now.

dvmarinoff, 2024-03-21:
@Garyrk001 Thanks for the report!

Just added version 0.1.43 here: [https://flux-stable.vercel.app/](https://flux-stable.vercel.app/).
This one is before the big Bluetooth changes. Meanwhile will try to fix Wahoo Kickr Snap for the latest version, and will let you know when there is some progress.

Garyrk001, 2024-03-21:
Thank you!
With 0.1.43, the trainer seems to work correctly.
Strangely, I am still having problems with the cadence sensor on 0.1.43.  I checked it on other apps, and it works correctly in those apps.  This is not a big deal, as I can run through workouts without cadence readings.

I went back to confirm 0.1.48 trainer issues, and ERG mode seems to have the issues.  The slope and resistance modes seemed ok when I manually change the level with the+/- buttons.  Just ERG mode doesn't do anything.

If you need any data or debugging info, let me know details and how to collect and I'll try to provide.

Thank you for the quick response.

dvmarinoff, 2024-03-21:
So I think I found the issue with the code for the Wahoo Kickr Snap. Power was not right, because I set the value to index 0 instead of index 1. This is now covered by the test cases, and I additionally tested with a Snap simulator. The simulator logs look good, but a test with the real device would be better.

When you have the time could you follow the instructions from this post and link the logs from the dev console here:
[how to get logs](https://github.com/dvmarinoff/Flux/issues/186#issuecomment-2009602109)
Just make sure to hard reload the page in order to get the latest version.

The cadence issue is probably related to localstorage, but the logs will show that if you can switch on and off cadence from different devices in settings during the test ride. If that's the case I'll address it in the next update.

Garyrk001, 2024-03-22:
Thank you again for looking into it.
On the logs, if I read the post you reference correctly, it is only possible on a PC/MAC?  Is there a way to get the logs on Android?  If not, it might take me a little longer to find a friend's laptop I can borrow to collect the logs.  If my friends are kind, I might be able to get to it in a few days.

dvmarinoff, 2024-03-22:
No worries, a simple test with the latest version on Android would do. Just to confirm it works.

Garyrk001, 2024-03-22:
I just did a workout with 0.1.48.1.
The trainer worked and was controlled as expected through the workout. This is great!

I then put on my QA hat and played with the+/- buttons. They worked, but we're off a little. For example, if I increased the target power to 220 in erg mode, the trainer seemed to keep me at 210.  Note that I never used this before so I do not know if it is new.  And honestly, I don't have a need for it. I just follow the workouts 

The cadence sensor still didn't work. It pairs, but I just get -- for my cadence. I'll play with it a bit more.

Thank you again for the help and quick turnaround on the trainer issue.

dvmarinoff, 2024-03-22:
Nice! Thanks for the test!

About the +/- buttons, the issue comes from the fact that the User can click the buttons faster than the trainer can do the power target updates.

In the old version I handled this with by denouncing the click function, but if the User clicks in a certain pattern they could still end up with the power target minus one increment, -10 watts in this case. 

I will be solving this issue, but I am still looking for an elegant way to approach it.

Garyrk001, 2024-05-30:
A little more of an update...  I have been using the app since March, and it is generally working well (Thank you!).  However, 3 different workouts, I have seen an issue where the workout is suggesting one power level (ERG mode), and then suddenly the trainer jumps to a much higher resistance/power level.

This only has happened on longer workouts (>45 minutes) as each time it occurred at 50-60 minutes in the workout.  I am typically doing a cool down by then, so the workout target power has been low - 80-100 watts and then suddenly the resistance jumps up and I need to put out ~200 watts.  The screen continues to show the 80-100 watts, but resistance jumps and requires me to put out the 200 watts.    If I hit the + or - keys, it seems to jump back down to the newly set power and continue.  

I have not done longer than 60 minute workouts, or continued for long periods of time after the power jump to see if/when it clears up.  Unfortunately, I can not see a clear pattern when it does this.  Just that it has been after at least 50 minutes.

Not the worst thing in the world, but I thought I would let you know.

dvmarinoff, 2024-05-30:
Hey, that's great thanks for the update!

I've been using the web app for my base training the last 3 months. My workouts are usually at least 90 mins, but the intervals are all z1 and z2, with very rare z4, no funky intervals or ramps.

What you describe may be caused by:
1. the workout file not parsing correct
2. the cooldown interval not reading correct
3. the current interval index jumping

Can you upload here or paste the contents of the files of some of the trouble making workouts?

Garyrk001, 2024-05-31:
Hi,  Thank you for looking into this.  Note that I have not repeated these workouts and I am not sure if the issue I saw would occur again.  It does seem a bit random.   But I attached a zip file with 2 workouts. One I am sure is correct, as I ran it today. That one is the "FTP Builder 8211 Week 5" file.  On that one today,   I was running at FTP 170 (I'm old, give me some slack :) ) and at 59:50 the power jumped from 85 watts (correct) to about 195.  Given there were only a few seconds left, I just finished it out.

The 2nd file is the "G1" file.  I believe that is another one I ran into the issue on.  But that was a month ago, and I'm trying to match up the power output in garmin connect with the workout file - so slight chance I'm getting it wrong.  But assuming it is correct - it looks to match up with my results, when running that file, FTP 165, at 57:00 the power jumped from 82 watts (correct) to about 200 watts.  After about a minute of that, I hit the "-" button a few times and it dropped to 72 watts, which was shown on the screen as the target.

Both of these issues happened in the middle of an interval.

[Problem workouts.zip](https://github.com/user-attachments/files/15506967/Problem.workouts.zip)

Garyrk001, 2024-06-06:
I repeated the FTP Builder 8211 Week 5 workout, and as I feared, it did not repeat the problem.  All worked as expected.
I don't know if there is anything in the app that could vary, is timing related, or even changes based on bt communications, etc.

Other options might be that my trainer gets wonky after a bit and overheats or something.  Maybe I'll try some longer workouts on different apps. I know I did some longer workouts months ago in another app and didn't have issues.  But that was then.

If I find out any more, I'll post.  Thanks for the support.

closed 2024-10-01 by dvmarinoff