dvmarinoff/Auuki #189: [Feature Request] - Ability to load a workout via a url parameter

daniellytle, 2024-04-17:
I'm working on a directory of workouts in .zwo format. I'd love to be able to link to https://flux-web.vercel.app/ with a zwo file encoded in the url and start using that workout.

Is this something you think would be valuable for the app?

dvmarinoff, 2024-04-19:
Not sure I follow, can you describe the use case further. Where is the bottleneck currently? The upload?

Also would integration with an existing platform like say intervals.icu or training peaks, help? I am not sure if those platforms have workout manager, but maybe some other platform out there has.

daniellytle, 2024-04-24:
Sorry that was very vague, I'll add some more context. I'd like more variability in the types of workouts I can quickly access in Flux. I started to build [cycling-workouts.com ](https://cycling-workouts.com) to host a large number of comparable workouts. I think it'd be a great flow to link from this workout site to Flux and be ready to ride that workout.

I'm curious if you would be open to building an integration into flux where a workout in cycling-workouts would have a "Open in Flux" button that would link to flux with that workout selected. Another option would be building cycling-workouts directly into Flux.

Let me know what you think about this! Currently I will download the workout directly and upload it to Flux and ride it there, this would be a nice integration to speed up that process.

dvmarinoff, 2024-04-24:
I'll think about it, maybe when I publish the Flux API, I could add an endpoint for syncing a workout, maybe some api auth, etc. What are your plans for `cycling-workouts.com`? Workout Editor?

daniellytle, 2024-04-30:
Sounds good, maybe it could be a database that Flux could draw from? I don't think I'll build an editor into it as there are a few that exist already and it's my thought that I'd rather do a workout made by a professional than build one myself. IMO the appeal is more "I'm gonna do a workout that the pros do".

I'll keep working on it as a database and circle back in a bit!

closed 2024-04-30 by daniellytle