dvmarinoff/Auuki #192: record HRV data from compatible devices

dvmarinoff, 2024-05-01:
- read from the Polar H10 strap
- record to .FIT file
- research the current state of DFA a1 estimate

fab33, 2025-01-16:
DFA alpha1 will be fine too. With same custom fields from Garmin AlphaHRV app.

dvmarinoff, 2025-01-30:
Posting here some notes I got from implementing this for iOS.

# Data
Here are some Data examples of full Heart Rate Measurement messages. These can become test cases:
```
[16, 67, 114, 3, 138, 3]
[16, 68, 146, 3]
[16, 60, 140, 4]
[0, 64]
```

# Spec
Here is the Hear Rate Measurement message as defined by the bluetooth spec:
 ## Field 0, Flags
 ```
 16
 0b00010000

 0 bit, HeartRateValueFormat, 0 -> UInt8, 1 -> UInt16
 1 bit, SensorContactStatus, 0 -> no contact, 1 -> contact
 2 bit, SensorContactSupport, 0 -> not suppoted, 1 -> supported
 3 bit, EnergyExpenditureStatus, 0 -> not present , 1 -> present
 4 bit, RRInterval, 0 -> not present , 1 -> present
```
 ## Field 1, Heart Rate

 ## Field 2, Energy Expenditure
 
 If present should be included once every 10 measurements
 If value is 0xFFFF the value needs to be reset by the client

 ## Field 3, RRInterval
 
 length is up to 8 if EnergyExpenditure is present
 lenght is up to 9 if EnergyExpenditure is not present
 Resolution of 1/1024 second

 HR = 60 / RR Interval in seconds

# FIT File
```
 HRV messages are used to record heart rate variability data. The HRV messages contain an array of RR
 intervals and are interleaved with record and event messages in chronological order. Note that HRV data is
 not timestamped, and shall be synchronized by checking successive RR intervals as they occur between other
 timestamped messages.
```

[image]

Inside the fit file hrv is not recorded as part of the record message, but as a separate message. But the HRV message does not have timestamp. It's timestamp must be derived from the previous message (preferably a record message). Another oddity is that the it has only one field which is fixed size array of length 5 which contains the RRIntervals, and is filled up with 0xFFFF where value is not available.

Therefore in the current implementation HRV messages must be recorded together with Record messages in `db.records`.
The FIT Parser will need to implement support for array type too.

Also note that while the Heart Rate Measurement message uses resolution of 1024 the FIT File format uses resolution of 1000.

EnricoFerro, 2025-01-30:
Ok, I'm working on this solution too...

I've created an implementation on  https://github.com/EnricoFerro/Flux/tree/hrv2-wip

It write something.. ...there only some problem with the empty fields:
[image]

dvmarinoff, 2025-01-30:
I am going to try to implement the FIT Parser logic in a more general way. Ideally the global definition files should be read only.
They are generated with a script from the official spec document.

mgibbs189, 2025-01-30:
@dvmarinoff fyi the `fit-javascript-sdk` now supports both encoding and decoding.

It's still quite a bit of extra overhead (namely profile.js), but might be a worthwhile trade-off...

dvmarinoff, 2025-01-31:
@mgibbs189 Hey, thanks for the update!

PR #241 (2025-02-01): add support for HRV recording
- add support for encoding and decoding array types in the fit parser
- add hrv message in the product message definitions
- add test cases that cover array types and hrv message
- fix some minor bugs with the fit parser

- add support for full decoding of the Heart Rate Measurement characteristic
- add test cases for decoding Heart Rate Measurement Data

- transmit `rrInterval` from `reactive-connectable` if present in HR data
- add support for recording RR Intervals in `db` as the `rrInterval` field
- add `rrInterval` to records

The RR Intervals array seems to  [... cut]
closed 2025-02-13 by dvmarinoff