dvmarinoff/Auuki #195: Text events from zwo files

Jinglessss, 2024-07-27:
I have a set of zwo files that I created in that popular subscription based training app, that include text events.  I can see this with time stamps in the files, is there a way these can be displayed on screen when using these workouts?

dvmarinoff, 2024-08-06:
Currently those are not supported, but since many people are requesting it, I can add support. Although I can't promise a timeline at the moment.

Jinglessss, 2024-08-06:
No problem, just an idea. So far I'm liking the site just for it's
lightweight simplicity, runs on any of my devices and is a lot less messing
around than dealing with a Zwift login to just follow erg mode.

richardhamwi, 2025-11-02:
Hey, I'm learning to code and I thought I would use Ai to try and build this feature. I've got a rough version working. Very buggy still but maybe there is some potential here.
I love your app!

[image]

dvmarinoff, 2026-10-19:
There is a test version of this feature on dev.auuki.com now. You can give it try and let me know if I have to fix or adjust something.

richardhamwi, 2026-10-19:
> 4\. There is a test version of this feature on dev.auuki.com now. You can give it try and let me know if I have to fix or adjust something.

I've just tested this. It works well. As a user, I would happy to use this feature. It adds a lot to the experience. Thanks

napfbike, 2026-10-24:
Have also checked and used the text events on dev. Looks good and would be great to have it.
Thanks!

dvmarinoff, 2026-10-26:
Thanks for testing the feature! I am adding it to the main version.

closed 2026-10-26 by dvmarinoff