dvmarinoff/Auuki #197: "Browser not supported" banner when self-hosted

andrewjamesturner0, 2024-08-10:
Thanks for the great project.

**I'm trying to run this on my own server. However, even in a supported browser (Chrome), I'm getting the 'This browser is NOT supported" banner.**
 
* I'm confident the browser is not the issue, because https://flux-web.vercel.app/ works perfectly in a different tab.

* I've tried with version 0.1.48.1 and 0.1.43: no difference.

* Only difference I can think of is that I'm not using SSL, since it's all local on my LAN. Does that matter?

**Steps to reproduce**

1. Create VM (I've been using ubuntu server 24.04)

2. Install nvm (curl | bash, :-/ I know, I know.)
```
$ curl -o- https://raw.githubusercontent.com/nvm-sh/nvm/v0.39.3/install.sh | bash
$ source ~/.bashrc
```

3. install node
```
$ nvm install lts/Iron
```

4. get Flux
```
$ git clone https://github.com/dvmarinoff/Flux.git
$ cd Flux/
$ npm install
$ npm run start
```

5. visit server from supported browser
```
<IP>:1234
```

dvmarinoff, 2024-08-14:
Hm, no idea about that. Web-bluetooth requires https, is all I can think of right now. I could try to reproduce the setup, maybe something will come up.

BTW if you could install the app on your device as a PWA, and have the ability to use it offline, would that help your use case?

andrewjamesturner0, 2024-08-16:
Thanks, I'm now pretty certain https is the issue. Setting up a local CA on my lan is not a particular problem, but I'm unsure how to give Flux the SSL bits. Can you give me some pointers on the proper way to deploy it? (I'm afraid I'm not at all familiar with parcel) Do I need to pass some extra arguments to parcel, or should I do it with an nginx reverse proxy?

closed 2024-09-26 by dvmarinoff