dvmarinoff/Auuki #198: Problem getting heart rate from heart rate monitor

FrederikThorsted, 2024-08-14:
After initial pairing of the sensor, the heart rate is shown on the dashboard. If I refresh the page, the sensor is still connected, but the dashboard is showing '0' under heart rate. 
The problem persists if I disconnect and reconnect the sensor. 

The heart rate is shown again if I remove and insert the sensor battery and reconnect. The problem comes back if I refresh the page again 

Heart rate device used: Garmin HRM Pro Plus

dvmarinoff, 2024-08-14:
Hi @FrederikThorsted  and welcome to the repo!

Thanks for the bug report!
The odd thing is that refreshing the page must lead to all devices disconnecting, and the workout pausing.
I tried various scenarios with my Polar H10 strap, but currently I can not reproduce the issue.

I’ve tested those scenarios:
- Connect from the main page, refresh, re-connect again
- Connect from settings page, refresh re-connect again
- Connect from settings page, refresh, re-connect from home page
- Repeat those with a workout started
- Repeat those with the HRM connected to another app (it supports dual connection)

OS: Mac
Browser: Google Chrome
HRM: Polar H10

https://github.com/user-attachments/assets/9d144db2-f89a-4cf9-8128-12fcee1aa168

Can you check if all is good with the device Settings and the HRM is selected as a source for Heart Rate and not some other device (the label should be white, not gray). Although this shouldn't be the case since removing the battery resolves it.

I'll further look into the Garmin HRM Pro Plus to see if it has something special about it from bluetooth connectivity standpoint.

FrederikThorsted, 2024-08-14:
Hi @dvmarinoff 

Sorry, I forgot to mention OS and browser. When I tried before, I used my phone:
Phone: Oneplus 8 Pro
OS: Android 13
Browser: Google Chrome 
HRM: Garmin HRM Pro Plus 

It seems to work fine when I try on my laptop
OS: Windows 10
Browser: Microsoft Edge 
HRM: Garmin HRM Pro Plus 

Attached screenshot is the settings page after page refresh and reconnected to the sensor:
[image]

dvmarinoff, 2024-08-14:
You are right, just ran some tests on Android. When refreshing the page with a bluetooth device connected both Google Chrome and the Default Android browser crash. On the other side Samsung Internet Browser does works as expected. It might be some web-bluetooth issue with Chrome, will keep digging.

Samsung Galaxy S9
Android 10
Google Chrome version: 127.0.6533.103

closed 2024-09-26 by dvmarinoff