dvmarinoff/Auuki #200: Simple slope based workout not working

NickTTT-alt, 2024-08-16:
Hello friend,

Firstly many thanks for your efforts on this fine app. I am hopeful I will be able to cancel my current subscription-based app soon and use this in its place, and will happily make a donation if I do so.

I would like to use simple slope based workouts with my trainer if possible, however so far this has been unsuccessful.

I create simple workouts such as this....

<workout_file>
    <author>Flux</author>
    <name>Simple Slope</name>
    <category>SweetSpot</category>
    <subcategory></subcategory>
    <description>Best workout ever!</description>
    <sporttype>bike</sporttype>
    <tags></tags>
    <workout>
        <SteadyState Duration="30" Power="0.5" Slope="2" />
        <SteadyState Duration="30" Power="0.7" Slope="4" />
    </workout>
</workout_file>

When I begin the workout the slope is set to 2 as expected, and the trainer's resistance corresponds to this, and all is great. After 30 seconds I would expect the slope to increase to 4, however this doesn't happen. And in fact the interval never seems to come to an end, with the interval time just going on and on.

Probably I am just misunderstanding the use of the app, but if you would be so kind as to take a quick glance as to my set-up and expectations I would be extremely grateful.

Many thanks.

dvmarinoff, 2024-08-16:
Hi @NickTTT-alt and welcome to the project!

> Is it reasonable to expect the app to automatically increase the slope

Yes the app should handle Power and Slope changes automatically. It’s odd that it doesn’t.

> Also the red line that marks where I am up to in the current workout starts mid-way along the power graph and travels from right to left

This app has 2 play buttons (I know a bit confusing). The one with the circle starts the workout and the recording, the plain one starts just the recording. Pressing just the recording play button, recreates this issue. The buggy progress line starts at the end of the first interval and oddly moves right to left, but pressing the workout start button fixes that. Also you’d see that the control mode automatically switches from ERG to Slope.

Now since I am currently on road I don’t have a trainer or an android phone with me and can’t fully test if everything is alright with Slope workouts, but ERG mode should be fine, I did a long workout yesterday with Chrome and Android. If you have any issues with trainer control you can checkout the old stable version here: https://flux-stable.vercel.app/

By next weekend I’ll back and will be able to check this more thoroughly. Just to make sure can you share more info about your trainer like Manufacturer, Model, Year?

About the red progress line, I know it has some issues, but I’ve been planing to change the entire code of the workout graph, so I am delaying this fix for later.

Here is a recording from a simulator that recreates the issue:
- timer going up, not counting down
- progress line in center
- progress line moves right to left
- automatic change to Slope mode not happing

https://github.com/user-attachments/assets/1c461934-b57e-4a1c-941c-585c27e14e89

NickTTT-alt, 2024-08-16:
It works!

It was entirely my fault using the wrong play button. Many apologies. I do wonder if this would possibly be made a bit more intuitive though?

I will start to transfer all of my workouts into Flux, and will look to setup my own local server. Very exciting.

There is just one metric I am interested in at the end of a workout and that is average power during each interval. Is this possible to extract?

Thanks again!

dvmarinoff, 2024-08-17:
> I do wonder if this could possibly be made a bit more intuitive though?

Yes, I’ve been wondering about that, maybe a tooltip, or interactive tutorial on first use, or a popup message, …

> average power during each interval

The app will record all data live in the browser, when you click pause, you’d be able to see the stop button, the stop button will end the workout and you’d have the download button available. If you click the download button it will export the data to a .fit file. This is the format used by most sports watches and cycling navigations (Garmin, Wahoo, etc.)

You can manually upload this file to a service like Strava or Garmin Connect, and those services will be able to decode the file and display the power averages. This app doesn’t calculate power averages for the intervals, but the data is available, and client apps can just calculate them from the interval start / end times and the records.

Here is a tool that displays what’s in the .fit file: https://www.fitfileviewer.com/

Here how things look in Strava and Garmin Connect:
[image]
[image]

NickTTT-alt, 2024-08-17:
That's fantastic, thank you. I have set up a Garmin Connect account and the .fit file upload works fine. At this point it seems that this upload is manual only correct? Or is there a way to automatically send each workout file at the cessation of the session?

A side note, in case anyone else has mentioned it also: I have had to adjust the slope on each of the workouts exported from my previous app to get the resistance required. For example, for zone II previously I used a slope of 2.9% which corresponded to ~145W power. With flux I had to change this to 2.1% to get the same resistance. If left at 2.9% the power would be way up at 180W! Strange right?

dvmarinoff, 2024-08-20:
Hey, 

sorry for the late reply been living on trains for the past few days, but here is the current state of things:

> At this point it seems that this upload is manual only correct

Yes, it is a manual upload only, at this point but things will change. This feature would require a server and a User login to be added. I have some of this ready, but I just need to decide on the way the login is going to be. I am looking to use passkeys, which are keeping the sensitive part of the authentication on the user’s side instead of in a remote database as is the case with the classic email + password login. In the future automatic upload will be possible, but at this point I am planning to support only Strava, Intervals.icu and Training Peaks. Garmin Connect does not allow direct upload anyway, you’d need to proxy this over Strava, TP or something else, it’s just that currently Garmin works with a very limited choice of partners.

On the different resistance issue. That’s to be expected since in order to determine the resistance the trainer uses not just slope, but also coefficient of rolling resistance (crr), wind speed, and wind resistance. There is a basic specification standard for each of those values, which Flux is following, but each app (Tacx, Zwift, Wahoo, etc.) is going to pick values they deem more suitable for their use case. So indeed setting 3.0 slope in Flux, is going to differ to setting 3.0 in Zwift, because they may have chosen different values for the rest of the parameters.

closed 2024-08-22 by dvmarinoff