dvmarinoff/Auuki #203: Button for skip not working

srnanek, 2024-09-30:
Hello, skip button not working I'm latest version but on 0.1.43 is working.

dvmarinoff, 2024-09-30:
Just ran a couple of tests with 0.1.49 on Mac and Android with Chrome. Can't seem to reproduce the issue. Both Lap/Skip button and the L shortcut key work there. We are talking about the button with the Watch Timer icon, right?

[image]

srnanek, 2024-09-30:
Yes the Wath timer icon. I had problem on Android 13 One UI 5.1 with app and also in Chrome browser. 
I used my own Workout but also a tested app default Workouts.

dvmarinoff, 2024-09-30:
My device is on Android 10, One UI 2.5. Can you attach some logs from the console as a screenshot or text.

srnanek, 2024-09-30:
I used it from debug window and now it works.
Srry and thnks for help

closed 2024-09-30 by srnanek