dvmarinoff/Auuki #205: Delete button not appears if too long word in workout name

karlacio, 2024-10-01:
Hi,

thanks for your great app!

I created a zwo file with a name containing a too long word.

Now in workout list the select button is shifted right and the three vertical dot button is no more visible.
It's not possible to delete it as you can see in the following picture:

[image]

dvmarinoff, 2024-10-02:
Hi, thanks for the reports!

I am noticing long titles are causing multiple issues, including with the graph titles, although I couldn't reproduce the delete button hiding. Can you share more info about the browser, os?

 
[image]

karlacio, 2024-10-02:
I'm using it on my Samsung A52s (screen size 1080 x 2400 pixels, 20:9 ratio ~405 ppi density), android 14.

It's Flux 0.1.49 and I'm using it on Chrome 19.0.6668.70 or in the app created from Chrome.

I suppose the problem is due to the long word "(8min@105,98,91)+4min@60" in the title that prevent wrapping.

I was able to delete the workout opening flux with chrome and switching to desktop view.

dvmarinoff, 2024-10-02:
Just pushed some fixed to the developer version: flux-dev.vercel.app

All long workout names and workout categories get truncated now in the menu, the workout list, and the workout graphs.
Used css-grid for those parts. In time I should transition everything to css-grid.

closed 2024-10-03 by dvmarinoff