dvmarinoff/Auuki #209: Can't connect BLE power/cadence sensor to Android using latest/0.1.60

jimdavcon, 2024-10-19:
I've been using Flux (app from Vercel) for months without difficulty with a homebrew BLE power sensor connecting to my Android phone running Chrome. A month or so ago, Flux stopped showing any power readings (i.e., '-' displayed, not even '0') although it seems to connect to the sensor successfully. When I switch to the 'old stable' version (0.1.43), the readings work again.

When I connect the sensor to the 0.1.60 version of Flux on a laptop (Windows 11, with Chrome), the readings work as well so I'm sure the sensor is still working. Seems like something about parsing messages from the cycling power service got broken?

dvmarinoff, 2024-10-19:
Hi and welcome to the project! Thanks for the report!

So the sensor works, but not on Android specifically with 0.1.60. Since the3 versions are different instance each one may have Settings in different state. Can you first confirm that the Setting to read power from the power meter is switched on.

Otherwise that's an Android issue. Might be difficult to figure out. BTW what is a 'homebrew BLE power sensor'? Did you build your own power meter?

jimdavcon, 2024-10-20:
I use an esp32 with crank and wheel speed sensors to populate the cps and ftms BLE messages. So I  compute a 'fake' power measurement but it had been working great for quite a while on Android (and the 0.1.43 version still does work). Something in 0.1.60 seems to no longer like those messages.

I'm not quite sure what else is necessary to turn on the power sensor in flux beyond connecting the BLE  sensor and tapping the power measurement to make it bold. I've done that and neither power or speed/cadence start updating.

jimdavcon, 2024-10-20:
Sorry, please disregard!  I think I was just missing when I thought I was tapping on the power sensor..  maybe it could change from white to green text when it's 'on'? In any case, user error, my apologies.

closed 2024-10-20 by jimdavcon

dvmarinoff, 2024-10-22:
Also nice work with esp32! You are giving me an idea. Maybe I could use one of those as a configurable device simulator for testing. Currently I am using nRF Connect on an android phone for simulating bluetooth devices, and this has some. drawbacks.