dvmarinoff/Auuki #212: Workout graph cut after screen rotation

karlacio, 2024-11-19:
Hi,

I usually use Auuki in with my phone placed in vertical.

Sometime I have screen rotation active and if I accidentally switch to lanscape mode, when I return back in vertical mode, the workout graph will be cut on the right side.

After the cut in landscape is ok but in vertical mode remain always cut (even if i load a new workout).

I attach some screenshot to show the problem:

[image]
[image]

Carlo.

dvmarinoff, 2024-11-20:
Thanks for the report, I've been noticing something like that too. In my case the graph won't adjust size when the window gets smaller width, but will adjust when the window gets larger width.

Anyway it turns out it's a one liner fix in the css. I will add it to the dev version, and later to master.

karlacio, 2024-11-21:
> In my case the graph won't adjust size when the window gets smaller width, but will adjust when the window gets larger width.

Yes, this should be the problem.

Thanks for your work!

Carlo.

commit 3abc89f3 (2024-11-20): fix #212 Workout cut after screen rotation

- set max-width based on vw for the workout graph
- extend the media query for landscape phone orientation from 428 to 432
closed 2024-11-27 by dvmarinoff